facebookresearch/habitat-lab #1445: X Error of failed request

taylorlulu, 2023-07-06:
I'm facing this problem when ran the interactive_play.py. And other examples work properly.
X Error of failed request:  BadAccess (attempt to access private resource denied)
  Major opcode of failed request:  151 (GLX)
  Minor opcode of failed request:  5 (X_GLXMakeCurrent)
  Serial number of failed request:  97
  Current serial number in output stream:  97
I've tried habitat 0.2.4 and habitat 0.2.3 howerver they reported the same error. How can I fix this problem?
I'm using 
[image]
I have no idea this problem was caused by the installtion of habitat or just a driver issue.
I would appreciate your help.

NPU-YanChi, 2023-07-06:
I also have same question

CallmeAG, 2023-08-09:
Did you solve it yet? I met the same problem and I don't know what to do with it

NPU-YanChi, 2023-08-09:
I change the version of habitat-sim and habitat-lab to v0.2.1, then solved. You can try more version.

CallmeAG, 2023-08-10:
Thanks for the reply. I followed your recommendation changing the habitat-sim and habitat-lab version to v0.2.1. When I install habitat-sim everything goes find.But I got this error when I Did
python setup.py deveplop --all
The exact error output:
Searching for tensorflow==1.13.1
Reading https://pypi.org/simple/tensorflow/
No local packages or working download links found for tensorflow==1.13.1
error: Could not find suitable distribution for Requirement.parse('tensorflow==1.13.1')
Did you have the same problem too?

NPU-YanChi, 2023-08-10:
maybe you can try other python version, such as 3.6

CallmeAG, 2023-08-11:
Finally I got no error running the commands trying python version 3.6 to build my conda env
So thankful for your help!

3050671462, 2023-12-10:
> Finally I got no error running the commands trying python version 3.6 to build my conda env So thankful for your help!

Could you help me? I also use python = 3.6 and habitat0.2.1(both lab and sim). But after I successfully install all the dependence, I run
python examples/interactive_play.py --never-end
 it failed and output:
UserWarning: WARN: Gym minimally supports python 3.6 as the python foundation not longer supports the version, please update your version to 3.7+
  "Gym minimally supports python 3.6 as the python foundation not longer supports the version, please update your version to 3.7+"
Traceback (most recent call last):
  File "examples/interactive_play.py", line 43, in <module>
    from habitat_baselines.utils.render_wrapper import overlay_frame
  File "/home/qikang/habitat-lab-0.2.1/habitat_baselines/__init__.py", line 7, in <module>
    from habitat_baselines.common.base_il_trainer import BaseILTrainer
  File "/home/qikang/habitat-lab-0.2.1/habitat_baselines/common/base_il_trainer.py", line 10, in <module>
    from habitat_baselines.common.base_trainer import BaseTrainer
  File "/home/qikang/habitat-lab-0.2.1/habitat_baselines/common/base_trainer.py", line 18, in <module>
    from habitat_baselines.common.tensorboard_utils import TensorboardWriter
  File "/home/qikang/habitat-lab-0.2.1/habitat_baselines/common/tensorboard_utils.py", line 11, in <module>
    from torch.utils.tensorboard import SummaryWriter
  File "/home/qikang/miniconda3/envs/habitat/lib/python3.6/site-packages/torch-1.10.2-py3.6-linux-x86_64.egg/torch/utils/tensorboard/__init__.py", line 4, in <module>
    LooseVersion = distutils.version.LooseVersion
AttributeError: module 'distutils' has no attribute 'version'
Have you met the same problem? How can I solve it?

EftePA, 2024-04-06:
This [suggestion](https://github.com/facebookresearch/habitat-lab/issues/1538#issuecomment-1902985545) solved the error for me, info: ubuntu 22.04, python 3.9. habitat-sim/lab/baselines: 0.3.1. Installed via conda.