facebookresearch/habitat-lab #1448: AttributeError: 'HabitatSim' object has no attribute '_initialized'

YYDS-cc, 2023-07-12:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.4

Habitat-Sim: v0.2.4 

## ❓ Questions and Help
Habitat-smi can run the Interactive testing with python：
"python examples/viewer.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb".

But when test the habitat-lab,
run    ./examples/shortest_path_follower_example.py
error: AttributeError: 'HabitatSim' object has no attribute '_initialized'.

Additionally, when run Non-interactive testing: python examples/example.py,
always download the ReplicaCAD dataset,
but I have downloaded the dataset offline and have changed the location of the dataset in the pick.yaml file.
Where should I place the dataset？

aclegg3, 2023-07-17:
Hey @YYDS-cc,

I'm not sure why you would get this error. Maybe your modules are not installed in the same environment or are otherwise not available when imported?

It may help to know more system details and to see log output.
How did you install the code?
Are you using a conda env?
What python version are you using?

etri-jeon0123, 2023-08-01:
I had the same problem, but it will be solved by installing a new conda and making the new environment the same.

In other words, I think the environment is updated frequently, whether it's a version or something.

aclegg3, 2023-08-17:
Thanks @etri-jeon0123 sounds like I can probably close this issue then.
@YYDS-cc if you have further issues you can open a new thread or re-open this if the solution provided is not sufficient.

closed 2023-08-17 by aclegg3