facebookresearch/habitat-lab #1487: Key 'action_space_config' not in 'SimulatorConfig'

etri-jeon0123, 2023-08-08:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.2.4
Habitat-Sim: v0.2.4

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/ : YES
Did you check out the tutorials? https://aihabitat.org/tutorial/2020/ : YES

## ❓ Questions and Help
Currently, we are testing code 4 of the challenge on the link below. 
(`https://github.com/facebookresearch/habitat-challenge`) An example on how to train DD-PPO model can be found in ~

There was no problem executing the code, 
but an error occurred due to the lack of **`"action_space_config"`** in the key "SimulatorConfig".

[image]

**code error**
[image]

`In 'benchmark/nav/objectnav/objectnav_v2_hm3d_stretch_challenge': ConfigKeyError raised while composing config:
Key 'action_space_config' not in 'SimulatorConfig'
    full_key: habitat.simulator.action_space_config
    reference_type=SimulatorConfig
    object_type=SimulatorConfig`

How do I fix an error caused by missing "action_space_config"?

aclegg3, 2023-08-08:
@ASzot @vincentpierre @rpartsey 
Any thoughts here?

kellyiss, 2024-07-04:
hi, have you solved this problem?