facebookresearch/habitat-lab #1491: Problem using the pretrained models in habitat-baselines. What configuration was used to obtain them?

CarlosSandoval03, 2023-08-09:
Hello,

I am trying to use the optimized pre-trained models available in habitat-baselines (https://dl.fbaipublicfiles.com/habitat/data/baselines/v1/habitat_baselines_v2.zip). https://github.com/facebookresearch/habitat-lab/blob/main/habitat-baselines/README.md

Specifically, I am trying to use the "gibson-rgbd-best.pth" model, which as far as I understand, uses RGB and Depth sensors, however, when trying to load the state dictionary, the dimensions and name of one of the networks does not match. The error is as follows:
_RuntimeError: Error(s) in loading state_dict for PointNavResNetPolicy:
Missing key(s) in state_dict: "net.pointgoal_embedding.weight", "net.pointgoal_embedding.bias". Unexpected key(s) in state_dict: "net.tgt_embeding.weight", "net.tgt_embeding.bias"._

If I just change the name from tgt_embeding to pointgoal_embedding then I get an error with the dimensions, given the amount of sensors used. As far as I understand, this is because I am using "pointgoal" sensor and the pre-trained model is using "IntegratedPointGoalGPSAndCompassSensor". 

**Thus, my problem is that I need a model optimized for navigation relying only on visual sensors (RGB and perhaps depth), but even though the name on the pretrained model file suggests so, when I load it looks like it was trained using also GPS and Compass sensors.** 

Hence, I would like to know if I could consult somewhere the config file used to obtain these pretrained models, or which configurations were used. If there is a model offered by habitat that is indeed optimized to navigate only using vision, I would also appreciate it.

dhruvbatra, 2023-08-09:
>  I would like to know if I could consult somewhere the config file used to obtain these pretrained models, or which configurations were used.

You should use these (which have the details you're looking for):  
https://github.com/facebookresearch/habitat-lab/tree/main/habitat-baselines/habitat_baselines/rl/ddppo

> If there is a model offered by habitat that is indeed optimized to navigate only using vision, I would also appreciate it.

https://github.com/rpartsey/pointgoal-navigation

closed 2023-08-09 by dhruvbatra

CarlosSandoval03, 2023-08-15:
Thank you for your answer @dhruvbatra 

I am trying to use the models you suggested (https://github.com/rpartsey/pointgoal-navigation) in the Pretrained Models - Navigation Policy section, but I am getting the following error:

AttributeError: Can't get attribute 'Config' on <module 'habitat.config.default' from '/scratch/big/home/carsan/Internship/PyCharm_projects/habitat-lab/habitat-lab/habitat/config/default.py'>

Do you know why it might be the case? I am only downloading the pth file and trying to load it with torch.load, nothing more. 
It works for my checkpoints and for the checkpoints in the Visual Odometry section, but not for those in Navigation Policy. 

Is it possible that the files in drive are damaged? Or, is the file working for you? 

Thank you all in advance.