facebookresearch/habitat-lab #1528: No visual results in the example.py. May lack of navmesh files.

Kairobo, 2023-08-23:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: stable
Habitat-Sim: following the conda install
Ubuntu 20.02

## 🐛 Bug

When I run the examples/example.py. The simulation window doesn't show up, and the following text is shown:
[image]

It seems some navmeshes are missed. But I think I have done the downloading part correctly.

Anyone can help me to check how to get those navmeshes?

closed 2023-08-30 by Kairobo

BenQLange, 2023-09-18:
How did you resolve it?

MeltedHyperion07, 2024-09-02:
can you elaborate on this a little. How do  i overcome this issue

Kairobo, 2024-09-09:
> @Kairobo can you elaborate on this a little. How do i overcome this issue

Just find the pictures you can use in the code and save it.