facebookresearch/habitat-lab #1533: what is the relationship between xxx.semantic.txt and xxx.semantic.glb?

zuixiaosanlang, 2023-08-24:
i get a rgb image from xxx.semantic.glb
[image]
and the floor RGB=(33, 185, 0)

but i cannot find a close color in xxx.semantic.txt about floor.

my code:
[image]

[image]

aclegg3, 2023-08-24:
Hey @zuixiaosanlang 
I believe your understanding is correct. The hex color in the text file should map to the color of the painted texture region in the .glb asset. We render a flat shaded image of that texture. 

If you can provide the specific scene name here we can take a closer look.

jturner65, 2023-08-24:
Hi @zuixiaosanlang
The GLAQ4DNUx5U.semantic.txt semantic text file should contain 907 entries, but it seems the file you are reading in only has 23.
[GLAQ4DNUx5U.semantic.txt](https://github.com/facebookresearch/habitat-lab/files/12430743/GLAQ4DNUx5U.semantic.txt)

zuixiaosanlang, 2023-08-25:
@jturner65 
Hi,
0x87E719 should be rgb : 135, 231, 25.    0x19=1*16+9=25

use MeshLab tool to compare:
[image]

[image]

[image]

it seems the floor color in your picture  is more closer to rgb(33, 185, 0)

zuixiaosanlang, 2023-08-28:
Hi  @aclegg3 
00861-GLAQ4DNUx5U

also you can check other scene, for example 00006-HkseAnWCgqk. it is the same problem. 

If it's my fault, please correct me.
thanks.

jturner65, 2023-08-28:
> @jturner65 Hi, 0x87E719 should be rgb : 135, 231, 25. 0x19=1*16+9=25
> 
Yes, that was a typo on my part, apologies, and thank you for correcting it.

Please let me know what mechanism you are using to query the color in your original post, so that as I attempt to figure out any potential issues I can follow the same pathway you did.

Thank you!

zuixiaosanlang, 2023-08-31:
@jturner65 
Hi,
i create tensor:
[image]

get rgb image from xxx.semantic.glb:
[image]

save rgb image:
[image]

then i check the floor color in rgb image, compared whith xxx.semantic.txt.

jturner65, 2023-08-31:
Are you setting up the sensor as a semantic sensor?

zuixiaosanlang, 2023-09-04:
Hi, @jturner65 
if i set sensor as a semantic sensor, i got all zeros.
[image]

then next line run failed
batch = batch_obs([observations])

[image]

[image]

zuixiaosanlang, 2023-09-04:
@jturner65 
i run the code and cannot get the semantic label.

https://aihabitat.org/docs/habitat-sim/image-extractor.html
[image]

[image]

and my env:
[image]

thanks.