facebookresearch/habitat-lab #1538: X Error of failed request:  BadAccess

YueFan1014, 2023-08-25:
Hi i'm using Ubuntu 20.04 and have the problem: 
`X Error of failed request:  BadAccess (attempt to access private resource denied)`
when running interactive play using command:
`python examples/interactive_play.py --never-end`.
I notice this problem has been already posted in README.md. I'm wondering whether this problem is later solved? Thanks!

lineojcd, 2023-08-25:
hi, I am using Ubuntu20.04 and I  have the same issue, and I wonder if there is any news for this issue. Thanks

bryanbocao, 2023-09-03:
Save error waiting to fix. :)

0mdc, 2023-09-10:
This is still an ongoing issue on Linux unfortunately.

Draculair, 2023-09-15:
I'm using ubuntu22.04 and I have the same problem

vdorbala, 2023-11-09:
I'm on Ubuntu 18.04 and am facing the same issue as well. So it doesn't seem to be OS specific.

Zeying-Gong, 2023-12-07:
It seems the bug still remains to be fixed... I'm using ubuntu22.04.

eder1234, 2023-12-14:
I'm facing the same issue in Ubuntu 18.04 after a local installation.

liuls2002, 2024-01-22:
I found out that this error occurs when executing the 476 line code : "pygame.init()". Moving this line of code somewhere else, it might work, such as moving to line 801, before starting the habitat.Env().

EftePA, 2024-04-06:
Thanks @liuls2002! Your suggestion worked for me!

waterproofship, 2024-06-14:
> I found out that this error occurs when executing the 476 line code : "pygame.init()". Moving this line of code somewhere else, it might work, such as moving to line 801, before starting the habitat.Env().

what do you mean? I can't get it to work

Sawyer-Wu, 2024-06-14:
您好，你的邮件我已经收到，我会尽快给您回复，谢谢。

clairetsai1222, 2025-01-03:
Thank you for your advice, @liuls2002! It worked for me!

Yukuncn, 2025-06-23:
> Thank you for your advice, [@liuls2002](https://github.com/liuls2002)! It worked for me!  Hello,I face the same issue,can you give me the detail of your code ?

yourkln, 2025-11-08:
could you please tell me, what did you do exactly: @Ivypool725