facebookresearch/habitat-lab #1544: How to set custom goal position/starting position for agent in Habitat lab?

azhannaqeeb, 2023-08-28:
I am new to habitat and I want to train an agent using RL in habitat lab. I tried already built in examples like shortest path follower but I am unable to find where I can change or provide starting/goal position for my agent. For each episode it starts from random position and ends at random goal. I want to find some configuration file where I can set these values for shortest path follower or any other script.

francescotaioli, 2023-08-31:
Tested for `habitat-sim` and `habitat-lab` 0.2.4
```
# import ...
scene = "cV4RVeZvu5T"
start_position = [6.25073, -0.881531, 9.50275]

config = habitat.get_config("config/point_nav.yaml")
with read_write(config):
    config.habitat.simulator.scene = os.path.join(config.habitat.dataset.scenes_dir, f"mp3d/{scene}/{scene}.glb")

rotation_quaternion = habitat_sim.utils.common.quat_from_angle_axis(np.pi, np.array([0, 1, 0]))

sim = make_sim(id_sim=config.habitat.simulator.type, config=config.habitat.simulator)

sim.set_agent_state(start_position, rotation_quaternion)
```

vantony1, 2024-04-29:
@francescotaioli is there a way to set a goal semantically? for example, set start location to be the kitchen and goal location to be the bedroom?

francescotaioli, 2024-04-30:
@vantony1, you can try to query the Simulator for the region and check the attributes. Maybe this could be useful: https://aihabitat.org/docs/habitat-sim/habitat_sim.scene.SemanticRegion.html