facebookresearch/habitat-lab #1595: ESP_CHECK failed, what am I supposed to do?

hbonnavaud, 2023-10-02:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: vx.x.x or master?

Habitat-Sim: vx.x.x or master?

Habitat is under active development, and we advise users to restrict themselves to stable releases of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases). The bug you are about to report may already be fixed in the latest version.

Master branch contains 'bleeding edge' code, but we do appreciate bug reports for it!

## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:

1. create a conda environment, and install habitat-sim and habitat-lab inside it.
2. Download the file at https://github.com/facebookresearch/habitat-lab/blob/main/examples/example.py
3. python3 example.py

```Traceback (most recent call last):
  File "/home/disc/h.bonnavaud/these/implems/one4all/test_habitat.py", line 25, in <module>
    example()
  File "/home/disc/h.bonnavaud/these/implems/one4all/test_habitat.py", line 9, in example
    with gym.make("HabitatRenderPick-v0") as env:
  File "/home/disc/h.bonnavaud/anaconda3/envs/habitat/lib/python3.9/site-packages/gym/envs/registration.py", line 676, in make
    return registry.make(id, **kwargs)
  File "/home/disc/h.bonnavaud/anaconda3/envs/habitat/lib/python3.9/site-packages/gym/envs/registration.py", line 520, in make
    return spec.make(**kwargs)
  File "/home/disc/h.bonnavaud/anaconda3/envs/habitat/lib/python3.9/site-packages/gym/envs/registration.py", line 140, in make
    env = cls(**_kwargs)
  File "/home/disc/h.bonnavaud/these/implems/one4all/habitat-lab/habitat-lab/habitat/gym/gym_definitions.py", line 92, in _make_habitat_gym_env
    env = make_gym_from_config(config)
  File "/home/disc/h.bonnavaud/these/implems/one4all/habitat-lab/habitat-lab/habitat/gym/gym_definitions.py", line 61, in make_gym_from_config
    return make_env_fn(env_class=env_class, config=config, dataset=dataset)
  File "/home/disc/h.bonnavaud/these/implems/one4all/habitat-lab/habitat-lab/habitat/utils/env_utils.py", line 37, in make_env_fn
    env = env_class(config=config, dataset=dataset)
  File "/home/disc/h.bonnavaud/these/implems/one4all/habitat-lab/habitat-lab/habitat/core/environments.py", line 126, in __init__
    base_env = RLTaskEnv(config=config, dataset=dataset)
  File "/home/disc/h.bonnavaud/these/implems/one4all/habitat-lab/habitat-lab/habitat/core/environments.py", line 46, in __init__
    super().__init__(config, dataset)
  File "/home/disc/h.bonnavaud/these/implems/one4all/habitat-lab/habitat-lab/habitat/core/env.py", line 382, in __init__
    self._env = Env(config, dataset)
  File "/home/disc/h.bonnavaud/these/implems/one4all/habitat-lab/habitat-lab/habitat/core/env.py", line 114, in __init__
    self._sim = make_sim(
  File "/home/disc/h.bonnavaud/these/implems/one4all/habitat-lab/habitat-lab/habitat/sims/registration.py", line 19, in make_sim
    return _sim(**kwargs)
  File "/home/disc/h.bonnavaud/these/implems/one4all/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 81, in __init__
    super().__init__(config)
  File "/home/disc/h.bonnavaud/these/implems/one4all/habitat-lab/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 294, in __init__
    super().__init__(self.sim_config)
  File "<attrs generated init habitat_sim.simulator.Simulator>", line 11, in __init__
  File "/home/disc/h.bonnavaud/anaconda3/envs/habitat/lib/python3.9/site-packages/habitat_sim-0.2.5-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 118, in __attrs_post_init__
    self.__set_from_config(self.config)
  File "/home/disc/h.bonnavaud/anaconda3/envs/habitat/lib/python3.9/site-packages/habitat_sim-0.2.5-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 227, in __set_from_config
    self._config_backend(config)
  File "/home/disc/h.bonnavaud/anaconda3/envs/habitat/lib/python3.9/site-packages/habitat_sim-0.2.5-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 200, in _config_backend
    super().__init__(config.sim_cfg, config.metadata_mediator)
AssertionError: ESP_CHECK failed: Error loading general mesh data from file data/replica_cad/configs/stages/../../stages/Stage_v3_sc2_staging.glb```

## Expected behavior

Example running.

## Additional context

Tested on ubuntu 20.04

Ask for any additional information.

aclegg3, 2023-10-02:
@hbonnavaud My first guess is that you don't have `git lfs` installed or there was an lfs error and your `.glb` files are actually pointer files, not the asset. Check that these file are actually glTF assets and not lfs pointers.

If they are pointers you should remove and re-download the dataset. Double check that `git lfs` is installed on your system beforehand.

hbonnavaud, 2023-10-02:
Hi,

It made the trick,
Thanks a lot !

Hedwin B.

closed 2023-10-02 by hbonnavaud

Yonggie, 2024-04-01:
Hi, what the dataset should be like? I've cheked the ``git lfs`` is installed.
## my hssd-hab dataset
[image]

## error

```
2024-04-01 09:37:23,896 Initializing dataset OVMMDataset-v0
2024-04-01 09:37:24,789 initializing sim OVMMSim-v0
Renderer: NVIDIA A100-SXM4-80GB/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 545.23.06
Using optional features:
    GL_ARB_vertex_array_object
    GL_ARB_separate_shader_objects
    GL_ARB_robustness
    GL_ARB_texture_storage
    GL_ARB_texture_view
    GL_ARB_framebuffer_no_attachments
    GL_ARB_invalidate_subdata
    GL_ARB_texture_storage_multisample
    GL_ARB_multi_bind
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_texture_filter_anisotropic
    GL_KHR_debug
    GL_KHR_parallel_shader_compile
    GL_NV_depth_buffer_float
Using driver workarounds:
    no-forward-compatible-core-context
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
Utility::Json: unexpected v at data/hssd-hab/stages/104348328_171513363.glb:1:1
Trade::GltfImporter::openData(): invalid JSON
```

Yonggie, 2024-04-01:
It seems that I missed to run ``git lfs pull``. And the errors went away after I did so according to this 
https://github.com/facebookresearch/home-robot/issues/347