facebookresearch/habitat-lab #1652: AssertionError: move_obj_reward measure requires object_to_goal_distance

Li-ChangHao, 2023-10-25:
When I use the command：

 `scripts/hab3_bench/bench_runner.sh`

I got an error: 

```
AssertionError: move_obj_reward measure requires object_to_goal_distance
                 listed in the measures list in the config.

```

Then I added object_to_goal_distance in habitat-lab/habitat-lab/habitat/config/benchmark/rearrange/hab3_bench/single_agent_bench.yaml

[image]

Then an error is reported: 

```
AssertionError: move_obj_reward measure requires ee_to_object_distance
                listed in the measures list in the config.
```
So I I added ee_to_object_distance in habitat-lab/habitat-lab/habitat/config/benchmark/rearrange/hab3_bench/single_agent_bench.yaml

Then got an error:

`hydra.errors.MissingConfigException: In 'benchmark/rearrange/hab3_bench/single_agent_bench': Could not load 'habitat/task/measurements/ee_to_object_distance'.`

Is there any solution?

closed 2023-10-27 by Li-ChangHao

duanyiqun, 2023-11-07:
Hi Changhao, I'm so excited you met the same problem with me. Have you solved the problem?
Any help would be very appreciated.

ruddradev, 2024-01-04:
Hi @duanyiqun : You need to change line 2247 in habitat-lab/habitat/config/default_structured_configs.py from - 
name="end_effector_to_object_distance"
to -
name="ee_to_object_distance"