facebookresearch/habitat-lab #1662: Interactive testing  fails on Ubuntu 20.04

JeffreyYH, 2023-10-31:
Note: Interactive testing currently fails on Ubuntu 20.04 with an error: X Error of failed request:  BadAccess (attempt to access private resource denied). We are working on fixing this, and will update instructions once we have a fix. The script works without errors on MacOS.

Looks like this problem has been here for months. Any plan to fix it?

zhibinQiu, 2023-12-01:
I'm also experiencing this issue on a Ubuntu 22.04 system.

liuls2002, 2024-01-21:
I found out that this error occurs when executing the 476 line code : "pygame.init()".  Moving this line of code somewhere else, it might work, such as moving to line 801, before starting the habitat.Env().

aclegg3, 2024-01-22:
Hey folks, sorry to hear about pygame troubles. We are working on deprecation pygame as an interactive platform. In meantime I suggest you try habitat-sim's viewer.py. You can import habitat-lab and modify the applications however you want.

jdiazram, 2024-01-31:
Hi everyone. I have the same problem in Ubuntu 22.04. Have any idea how to solve it? These are the last lines in my terminal:
```
MeshTools::compile(): ignoring Trade::MeshAttribute::TextureCoordinates 1 as its biding slot is already occupied by Trade::MeshAttribute::TextureCoordinates 0
MeshTools::compile(): ignoring Trade::MeshAttribute::TextureCoordinates 1 as its biding slot is already occupied by Trade::MeshAttribute::TextureCoordinates 0
MeshTools::compile(): ignoring Trade::MeshAttribute::TextureCoordinates 1 as its biding slot is already occupied by Trade::MeshAttribute::TextureCoordinates 0
MeshTools::compile(): ignoring Trade::MeshAttribute::TextureCoordinates 1 as its biding slot is already occupied by Trade::MeshAttribute::TextureCoordinates 0
[15:26:05:862971]:[Warning]:[Sim] Simulator.cpp(533)::instanceStageForSceneAttributes : The active scene does not contain semantic annotations : activeSemanticSceneID_ = 0
2024-01-31 15:26:07,308 Initializing task RearrangeEmptyTask-v0
b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,467]:
Bad inertia tensor properties, setting inertia to zero for link: r_gripper_finger_link
b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,467]:
Bad inertia tensor properties, setting inertia to zero for link: l_gripper_finger_link
b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,467]:
Bad inertia tensor properties, setting inertia to zero for link: r_gripper_finger_link
b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,467]:
Bad inertia tensor properties, setting inertia to zero for link: l_gripper_finger_link
X Error of failed request:  BadAccess (attempt to access private resource denied)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  5 (X_GLXMakeCurrent)
  Serial number of failed request:  89
  Current serial number in output stream:  89
```

feliyur, 2024-02-17:
Behavior seems related for me to the `SDL_VIDEODRIVER` setting used by pygame. The above happens for me when `SDL_VIDEODRIVER` is set to 'x11' (`export SDL_VIDEODRIVER='x11'`). When using 'dummy' the error appears to go away (but there's no display either, as expected), [other settings](https://www.pygame.org/docs/ref/display.html#pygame.display.init) for `SDL_VIDEODRIVER` do not work for me (a "not available" message"), possibly because I'm running in docker. Using `xhost` to give access to the root user does not change the BadAccess error.

Sawyer-Wu, 2024-02-17:
您好，你的邮件我已经收到，我会尽快给您回复，谢谢。

M-A-S1, 2025-06-17:
If anyone modified their viewer.py to enable interactions and robot agent in the environment then please share the modified script.