facebookresearch/habitat-lab #1672: Segmentation fault (core dumped) when testing habitat-sim

Wzey2000, 2023-11-06:
## version
System: docker with **Ubuntu 16.04.7 LTS**

Habitat-Sim: **v0.1.7, v0.3.0**

Driver: NVIDIA-Linux-x86_64-**470.57.02**

## ERROR

I ran habitat-sim v0.1.7 and habitat-lab v0.1.7 successfully before on the same server, but the server was accidentally reset, so I tried to reinstall habitat-sim v0.1.7 again. 

It came out Segmentation fault (core dumped) when testing using examply.py with Non-interactive testing.

```bash
python examples/example.py --scene data/scene_datasets/mp3d/1LXtFkjw3qL/1LXtFkjw3qL.glb
```

I tried installing habitat-sim v0.1.7 using conda and building from source, both of them could install successfully but came out the same error when testing:

```bash 
sim_cfg.physics_config_file = ./data/default.physics_config.json
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
Segmentation fault (core dumped)
 ```

I reset the 
```bash 
export GLOG_minloglevel=2
export MAGNUM_LOG=quiet
export HABITAT_SIM_LOG=quiet
```
to get the details, then it came out:

``` bash 
python examples/example.py --scene data/scene_datasets/mp3d/1LXtFkjw3qL/1LXtFkjw3qL.glb
sim_cfg.physics_config_file = ./data/default.physics_config.json
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1106 18:21:15.581204 26999 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : default changed to proposed JSON configuration filename : default.scene_dataset_config.json
I1106 18:21:15.581239 26999 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Dataset) : Proposing JSON name : default.scene_dataset_config.json from original name : default | This file  does not exist.
I1106 18:21:15.581354 26999 AssetAttributesManager.cpp:117] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I1106 18:21:15.581424 26999 AssetAttributesManager.cpp:117] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I1106 18:21:15.581490 26999 AssetAttributesManager.cpp:117] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I1106 18:21:15.581532 26999 AssetAttributesManager.cpp:117] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I1106 18:21:15.581555 26999 AssetAttributesManager.cpp:117] Asset attributes (cubeSolid : cubeSolid) created and registered.
I1106 18:21:15.581573 26999 AssetAttributesManager.cpp:117] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I1106 18:21:15.581629 26999 AssetAttributesManager.cpp:117] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I1106 18:21:15.581678 26999 AssetAttributesManager.cpp:117] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I1106 18:21:15.581708 26999 AssetAttributesManager.cpp:117] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I1106 18:21:15.581733 26999 AssetAttributesManager.cpp:117] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I1106 18:21:15.581769 26999 AssetAttributesManager.cpp:117] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I1106 18:21:15.581804 26999 AssetAttributesManager.cpp:117] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I1106 18:21:15.581821 26999 AssetAttributesManager.cpp:105] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
I1106 18:21:15.582878 26999 SceneDatasetAttributesManager.cpp:23] File (default) not found, so new default dataset attributes created and registered.
I1106 18:21:15.582890 26999 MetadataMediator.cpp:47] MetadataMediator::createDataset : Dataset default successfully created.
I1106 18:21:15.651480 26999 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Physics Manager) : Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json | This file  exists.
I1106 18:21:15.652289 26999 PhysicsAttributesManager.cpp:27] JSON Configuration File (./data/default.physics_config.json) based physics manager attributes created and registered.
I1106 18:21:15.652305 26999 AbstractObjectAttributesManagerBase.h:175] AbstractObjectAttributesManager<T>::createObject  (Stage) : Making attributes with handle : /root/qiaohongnew/zeyu_wang/ETPNAV/ETPNav/data/scene_datasets/mp3d/1LXtFkjw3qL/1LXtFkjw3qL.glb
I1106 18:21:15.652316 26999 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : /root/qiaohongnew/zeyu_wang/ETPNAV/ETPNav/data/scene_datasets/mp3d/1LXtFkjw3qL/1LXtFkjw3qL.glb changed to proposed JSON configuration filename : /root/qiaohongnew/zeyu_wang/ETPNAV/ETPNav/data/scene_datasets/mp3d/1LXtFkjw3qL/1LXtFkjw3qL.stage_config.json
I1106 18:21:15.652340 26999 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Stage) : Proposing JSON name : /root/qiaohongnew/zeyu_wang/ETPNAV/ETPNav/data/scene_datasets/mp3d/1LXtFkjw3qL/1LXtFkjw3qL.stage_config.json from original name : /root/qiaohongnew/zeyu_wang/ETPNAV/ETPNav/data/scene_datasets/mp3d/1LXtFkjw3qL/1LXtFkjw3qL.glb | This file  does not exist.
I1106 18:21:15.652529 26999 AbstractObjectAttributesManagerBase.h:181] AbstractObjectAttributesManager<T>::createObject  (Stage) : Done making attributes with handle : /root/qiaohongnew/zeyu_wang/ETPNAV/ETPNav/data/scene_datasets/mp3d/1LXtFkjw3qL/1LXtFkjw3qL.glb
I1106 18:21:15.652539 26999 AbstractObjectAttributesManagerBase.h:188] File (/root/qiaohongnew/zeyu_wang/ETPNAV/ETPNav/data/scene_datasets/mp3d/1LXtFkjw3qL/1LXtFkjw3qL.glb) exists but is not a recognized config filename extension, so new default Stage attributes created and registered.
I1106 18:21:15.652559 26999 Simulator.cpp:156] Loading navmesh from /root/qiaohongnew/zeyu_wang/ETPNAV/ETPNav/data/scene_datasets/mp3d/1LXtFkjw3qL/1LXtFkjw3qL.navmesh
I1106 18:21:15.653893 26999 Simulator.cpp:158] Loaded.
I1106 18:21:15.653921 26999 SceneGraph.h:93] Created DrawableGroup: 
Segmentation fault (core dumped)
```

I also debugged the example.py, the error appeared when it ran the line
```python  
super().__init__(config.sim_cfg)
```
in
```python 
def _config_backend(self, config: Configuration) -> None:
    if not self._initialized:
        super().__init__(config.sim_cfg)        # ERROR !!!
        self._initialized = True
    else:
        super().reconfigure(config.sim_cfg)
``` 
which is in /path/to/anaconda/env//lib/python3.6/site-packages/habitat_sim-0.1.7-py3.6-linux-x86_64.egg/habitat_sim/simulator.py.

I also tried the latest version of habitat-sim, v0.3.0, using conda install. When testing, the same error came out again:

```bash 
sim_cfg.physics_config_file = data/default.physics_config.json
Segmentation fault (core dumped)
```

## By the way

 The first bug I came across was the lack of libEGL.so etc, I solved these problems using [#1511 ](https://github.com/facebookresearch/habitat-sim/issues/1511) as I did not have the permission to reinstall the driver: 

```bash 
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/NVIDIA-Linux-x86_64-<version-id>
```

I dont know if it would cause the problem of Segmentation fault (core dumped).

## ❓ Questions and Help

I think this may be the problem of my server, because I ran successfully before.

Could you help me to identify the problem of my server or the direction to debug please?

xwjim, 2024-12-08:
Is this problem solved? I had the same problem

Wzey2000, 2024-12-11:
The problem is not solved yet.
---- Replied Message ----
***@***.***>Date12/8/2024 ***@***.***>***@***.***>,
***@***.***>SubjectRe: [facebookresearch/habitat-lab] Segmentation fault (core dumped) when testing habitat-sim (Issue #1672)
Is this problem solved? I had the same problem

Wzey2000, 2024-12-20:
> Is this problem solved? I had the same problem

Trying this may probably solve the problem:
https://github.com/facebookresearch/habitat-sim/issues/1511#issuecomment-1715699500

xwjim, 2024-12-21:
This works well for me. Thanks a lot.

zillur01, 2025-12-11:
tried the above solution, not working