facebookresearch/habitat-lab #1677: Issue with Downloading and Accessing the HSSD/hssd-hab Dataset

Jinghan11, 2023-11-09:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: vx.x.x or master?

Habitat-Sim: vx.x.x or master?

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
Hello,

I've encountered some trouble while attempting to download and utilize the HSSD/hssd-hab dataset from the Hugging Face [repository](https://huggingface.co/datasets/hssd/hssd-hab).

Initially, I tried to clone the dataset using the command `git clone https://huggingface.co/datasets/hssd/hssd-hab`, which executed correctly. However, I ran into permission issues when using the `python -m habitat_sim.utils.datasets_download --uids hssd-hab` command to download the dataset. I suspect that the datasets_download script might be utilizing SSH instead of HTTPS, which could be the root cause of the problem.

This issue seems to be causing further problems when trying to run the `python habitat_baselines/run.py --config-name=social_nav/social_nav.yaml` command. Despite being confident that I have the necessary file, I'm receiving a `FileNotFoundError: Could not find dataset file data/datasets/hssd/rearrange/train.`

Any help or guidance would be greatly appreciated. Thanks in advance.

Jinghan11, 2023-11-09:
The full traceback is as follows:
```
(habitat_0.3.0) ljh@ljh:~/habitat-lab/habitat-baselines$ python habitat_baselines/run.py --config-name=social_nav/social_nav.yaml
pybullet build time: Nov  2 2023 13:55:14
/home/ljh/miniconda3/envs/habitat_0.3.0/lib/python3.9/site-packages/torch/cuda/__init__.py:138: UserWarning: CUDA initialization: The NVIDIA driver on your system is too old (found version 11040). Please update your GPU driver by downloading and installing a new version from the URL: http://www.nvidia.com/Download/index.aspx Alternatively, go to: https://pytorch.org to install a PyTorch version that has been compiled with your version of the CUDA driver. (Triggered internally at ../c10/cuda/CUDAFunctions.cpp:108.)
  return torch._C._cuda_getDeviceCount() > 0
2023-11-09 11:25:27,894 Initializing dataset RearrangeDataset-v0
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/ljh/habitat-lab/habitat-baselines/habitat_baselines/run.py", line 77, in <module>
    main()
  File "/home/ljh/miniconda3/envs/habitat_0.3.0/lib/python3.9/site-packages/hydra/main.py", line 94, in decorated_main
    _run_hydra(
  File "/home/ljh/miniconda3/envs/habitat_0.3.0/lib/python3.9/site-packages/hydra/_internal/utils.py", line 394, in _run_hydra
    _run_app(
  File "/home/ljh/miniconda3/envs/habitat_0.3.0/lib/python3.9/site-packages/hydra/_internal/utils.py", line 457, in _run_app
    run_and_report(
  File "/home/ljh/miniconda3/envs/habitat_0.3.0/lib/python3.9/site-packages/hydra/_internal/utils.py", line 223, in run_and_report
    raise ex
  File "/home/ljh/miniconda3/envs/habitat_0.3.0/lib/python3.9/site-packages/hydra/_internal/utils.py", line 220, in run_and_report
    return func()
  File "/home/ljh/miniconda3/envs/habitat_0.3.0/lib/python3.9/site-packages/hydra/_internal/utils.py", line 458, in <lambda>
    lambda: hydra.run(
  File "/home/ljh/miniconda3/envs/habitat_0.3.0/lib/python3.9/site-packages/hydra/_internal/hydra.py", line 132, in run
    _ = ret.return_value
  File "/home/ljh/miniconda3/envs/habitat_0.3.0/lib/python3.9/site-packages/hydra/core/utils.py", line 260, in return_value
    raise self._return_value
  File "/home/ljh/miniconda3/envs/habitat_0.3.0/lib/python3.9/site-packages/hydra/core/utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
  File "/home/ljh/habitat-lab/habitat-baselines/habitat_baselines/run.py", line 31, in main
    execute_exp(cfg, "eval" if cfg.habitat_baselines.evaluate else "train")
  File "/home/ljh/habitat-lab/habitat-baselines/habitat_baselines/run.py", line 60, in execute_exp
    trainer.train()
  File "/home/ljh/miniconda3/envs/habitat_0.3.0/lib/python3.9/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/ljh/habitat-lab/habitat-baselines/habitat_baselines/rl/ppo/ppo_trainer.py", line 664, in train
    self._init_train(resume_state)
  File "/home/ljh/habitat-lab/habitat-baselines/habitat_baselines/rl/ppo/ppo_trainer.py", line 245, in _init_train
    self._init_envs()
  File "/home/ljh/habitat-lab/habitat-baselines/habitat_baselines/rl/ppo/ppo_trainer.py", line 143, in _init_envs
    self.envs = env_factory.construct_envs(
  File "/home/ljh/habitat-lab/habitat-baselines/habitat_baselines/common/habitat_env_factory.py", line 36, in construct_envs
    scenes = dataset.get_scenes_to_load(config.habitat.dataset)
  File "/home/ljh/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py", line 52, in get_scenes_to_load
    raise FileNotFoundError(
FileNotFoundError: Could not find dataset file `data/datasets/hssd/rearrange/train`
```

aclegg3, 2023-11-09:
@Jinghan11 this is failing because you don't have RearrangeEpisodes for HSSD. These are not the assets from the HSSD download, but rather episodes for rearrangement referencing the assets.

@ASzot do you have a pointer for episode download or do we expect users to generate their own datasets?

aclegg3, 2023-11-09:
Actually, there should be social rearrange episodes here (https://huggingface.co/datasets/ai-habitat/hab3_episodes) which can be aquired from the downloader with uid (hab3-episodes)[https://github.com/facebookresearch/habitat-sim/blob/22a95027893dd9155dd6921d7290cfbf219619bb/src_python/habitat_sim/utils/datasets_download.py#L76C9-L81C11]

Jinghan11, 2023-11-10:
Hi @aclegg3 , thanks for your explanation.

I've successfully executed the command `python -m habitat_sim.utils.datasets_download --uids hab3-episodes habitat_humanoids hab3_bench_assets,` yet the issue persists.

Do you have any ideas to debug this?

Jinghan11, 2023-11-21:
This issue is due to the wrong path of hssd-hab, not hab3-episodes.

closed 2023-11-21 by Jinghan11