facebookresearch/habitat-lab #1696: Crowd navigation and simulation with multiple humanoids with Habitat 3.0

rosanom, 2023-11-23:
Hello guys, first of all congrats for the great work done on the Habitat project!

I'm working on the crowd navigation problem (also referred as Social Navigation, but different from the social nav task defined in the Habitat 3.0 paper), where an agent has to reach a PointGoal while avoiding collisions with moving humans. I took a look at the `social_nav/social_nav.yaml` file, where I understand that a humanoid moves along shortest paths to randomly sampled coordinates, while the robot has to reach (and eventually follow) the humanoid, so it receives a HumanPointGoal as input, that is constantly updated. In my case, I would like to maintain the humanoid moving in the environment, but provide fixed PointGoals to the robot, like in a classic PointGoal navigation task. Is there a way to customize the new`social_nav.yaml` config file to support this new task?
Follow up question: can I import more than one humanoid to simulate very crowded scenarios?

Thank you in advance!
Marco

santurini, 2023-12-04:
Same problem here!

L-Scofano, 2023-12-04:
I'm also curious to know whether importing more than one humanoid is possible. Thanks in advance!

aclegg3, 2023-12-04:
Hey all,

Happy to see so much interest in socially aware AI applications!

It seems like the primary question here is around customization of Habitat 3.0 tasks with humanoids. Quick answers:

> Can I import more than one humanoid?

Absolutely! Habitat-sim has no limit to the number of articulated objects you can add to a scene. The humanoid_agent wrappers in Habitat-lab provide APIs to control the humanoid with animated walk cycles, root offsets, and other convenience utilities. These can also be instantiated in isolation from a task and are not limited to a single instantiation.

> How can I customize the new social tasks (e.g. social_nav)?

This will involve a combination of new task logic and configuration changes. I suggest you read into the task code, familiarize yourself with the details and then consider the following question: do you need to control your humanoid agents with policy actions or scripted control (e.g. shortest path following)? 

- If the former, you should pursue a multi-agent task with action inputs split between the agents in the scene. This will be a bit more complex.
- If the latter, you can write all humanoid logic in the task initialization and stepping functions. This is pretty straightforward. For example, think about starting from a point-nav task and simply implementing humanoid agents initialization and their crowd logic as additional dynamics in the task stepping phase.

xavierpuigf, 2023-12-20:
To follow up on the previous response. The more straightforward approach to add multiple humanoids to the social navigation task is to modify the `social_nav.yaml` file. You will need to do the following:

Right now there is agent_0 (robot) and agent_1 (human). You need to define a set of agents (say agent_2...agent_N) which will all be human, and have the same config as agent_1.

1. Go [here](https://github.com/facebookresearch/habitat-lab/blob/main/habitat-lab/habitat/config/benchmark/multi_agent/hssd_spot_human_social_nav.yaml#L10) and define the next human agents. You need to pretty much copy the config in agent_1 and make it for agent_2...agent_N (agent action, type, obs_keys, agents_order...)
2. Go then to `social_nav.yaml` and do the same, wherever there is an agent_1 specification, add for the agents_{2..N) you defined. 

With this setup you should be able to have something running.

rosanom, 2024-01-12:
Thank you for your answers! Now I better understand how to include multiple humanoids in my simulations. Do you have any suggestion on how I can fix the goal coordinates of the robot (not the humanoid) like in a standard pointgoal nav setup? I do not want the robot to reach the humanoid's position. 

> I would like to maintain the humanoid moving in the environment, but provide fixed PointGoals to the robot, like in a classic PointGoal navigation task. Is there a way to customize the newsocial_nav.yaml config file to support this new task?

Thank you