facebookresearch/habitat-lab #1697: Missing 'init_distributed' Function in the MultiUpdater Class Preventing Multi-GPU Training

Jinghan11, 2023-11-24:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.3.0

Habitat-Sim: v0.3.0

## 🐛 Bug

When attempting to perform multi-GPU training using the single_node.sh script, an error occurs due to the absence of the init_distributed function within the MultiUpdater class.

Upon reviewing the MultiUpdater class definition in [pop_play_wrappers.py](https://github.com/facebookresearch/habitat-lab/blob/main/habitat-baselines/habitat_baselines/rl/multi_agent/pop_play_wrappers.py#L551-L572), it is clear that the init_distributed function definition is indeed missing. However, this function is called upon in [ppo_trainer.py](https://github.com/facebookresearch/habitat-lab/blob/main/habitat-baselines/habitat_baselines/rl/ppo/ppo_trainer.py#L258).

## Steps to Reproduce

Run the following command:

```bash habitat-baselines/habitat_baselines/rl/ddppo/single_node.sh```

The output is :
```
Traceback (most recent call last):                                                                                                
  File "/home/user/habitat-lab/habitat-baselines/habitat_baselines/run.py", line 31, in main                                       
    execute_exp(cfg, "eval" if cfg.habitat_baselines.evaluate else "train")                                                       
  File "/home/user/habitat-lab/habitat-baselines/habitat_baselines/run.py", line 60, in execute_exp                                
    trainer.train()                                                                                                               
  File "/home/user/miniconda3/envs/habitat/lib/python3.9/contextlib.py", line 79, in inner                                         
    return func(*args, **kwds)                                                                                                    
  File "/home/user/habitat-lab/habitat-baselines/habitat_baselines/rl/ppo/ppo_trainer.py", line 664, in train                      
    self._init_train(resume_state)                                                                                                
  File "/home/user/habitat-lab/habitat-baselines/habitat_baselines/rl/ppo/ppo_trainer.py", line 258, in _init_train                
    self._agent.updater.init_distributed(find_unused_params=False)  # type: ignore                                                
AttributeError: 'MultiUpdater' object has no attribute 'init_distributed' 
```

## Additional context

This error indicates that the 'MultiUpdater' object does not have the 'init_distributed' attribute, confirming the function's omission within the class. This issue hinders multi-GPU training operations.·

0mdc, 2023-11-24:
@ASzot Would you have some context on this?

L-Scofano, 2024-01-17:
I'm experiencing a similar problem. Have you managed to find a solution? Thanks in advance.

neale, 2024-01-17:
Experiencing this issue too with multigpu training, would love some help.

ajperry2, 2024-01-17:
Think I found the issue that is calling this. The `init_distributed` function is a member of the `self._agent` object. The line in `ppo_trainer.py` should be `self._agent.init_distributed(find_unused_params=False)  # type: ignore `

As a quick fix in my entry point I substitute this line automatically with the sed command like so:
`sed -i 's/self._agent.updater.init_distributed(find_unused_params=False)  # type: ignore/self._agent.init_distributed(find_unused_params=False)  # type: ignore/g' habitat-baselines/habitat_baselines/rl/ppo/ppo_trainer.py`

Just make sure you are in the appropriate directory for the relative path. You can of course edit this manually if you are running on the host (I'm running in a docker).

Jinghan11, 2024-02-05:
This bug has been resolved in the following pull request: [Spot social nav #1658](https://github.com/facebookresearch/habitat-lab/pull/1658#issue-1969010767).

closed 2024-02-05 by Jinghan11