facebookresearch/habitat-lab #1701: Does the latest version of habitat sim support EquirectangularDepthSensor?

zhai-create, 2023-11-28:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.3.0 
Habitat-Sim: v0.3.0 

I want to use EquirectangularDepthSensor data in the experiment, but it seems that I cannot successfully complete the configuration. May I ask if the latest version of habitat sim supports EquirectangularDepthSensor?
(EquirectangularDepthSensor seems to be supported on previous versions (such as v0.2.0) of habitat lab and habitat sim.)

## ❓ Questions and Help

aclegg3, 2023-11-28:
@zhai-create yes, the EquirectangularDepthSensor should be fully supported. If you are encountering an issue feel free to post more details and we can help look into it.

zhai-create, 2023-12-23:
> @zhai-create yes, the EquirectangularDepthSensor should be fully supported. If you are encountering an issue feel free to post more details and we can help look into it.

Can you specify the specific configuration steps? I seem unable to complete the configuration of EquirectangularDepthSensor in the yaml file.

zhai-create, 2023-12-23:
> @zhai-create yes, the EquirectangularDepthSensor should be fully supported. If you are encountering an issue feel free to post more details and we can help look into it.

I configure it in the yaml file using the following method:
simulator:
    agents:
      main_agent:
        sim_sensors:
          rgb_sensor:
            width: 512
            height: 512
          equirect_depth_sensor:
            width: 481
            height: 241
    habitat_sim_v0:
       gpu_device_id: 0
But when the code runs, it will generate the following error message:
omegaconf.errors.MissingMandatoryValue: Missing mandatory value: habitat.simulator.agents.main_agent.sim_sensors.equirect_depth_sensor.type
    full_key: habitat.simulator.agents.main_agent.sim_sensors.equirect_depth_sensor.type
    reference_type=SimulatorSensorConfig
    object_type=SimulatorSensorConfig
Exception ignored in: <function Simulator.__del__ at 0x7f7bab357c10>
Traceback (most recent call last):
  File "/home/zhaishichao/anaconda3/envs/habitat4/lib/python3.9/site-packages/habitat_sim-0.3.0-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 528, in __del__
  File "/home/zhaishichao/anaconda3/envs/habitat4/lib/python3.9/site-packages/habitat_sim-0.3.0-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 131, in close
AttributeError: 'HabitatSim' object has no attribute '_initialized'

zhai-create, 2023-12-23:
> @zhai-create yes, the EquirectangularDepthSensor should be fully supported. If you are encountering an issue feel free to post more details and we can help look into it.

I configure it in the yaml file using the following method:
```
simulator:
agents:
main_agent:
sim_sensors:
rgb_sensor:
width: 512
height: 512
equirect_depth_sensor:
width: 481
height: 241
habitat_sim_v0:
gpu_device_id: 0
```
But when the code runs, it will generate the following error message:
```
omegaconf.errors.MissingMandatoryValue: Missing mandatory value: habitat.simulator.agents.main_agent.sim_sensors.equirect_depth_sensor.type
full_key: habitat.simulator.agents.main_agent.sim_sensors.equirect_depth_sensor.type
reference_type=SimulatorSensorConfig
object_type=SimulatorSensorConfig
Exception ignored in: <function Simulator.del at 0x7f7bab357c10>
Traceback (most recent call last):
File "/home/zhaishichao/anaconda3/envs/habitat4/lib/python3.9/site-packages/habitat_sim-0.3.0-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 528, in del
File "/home/zhaishichao/anaconda3/envs/habitat4/lib/python3.9/site-packages/habitat_sim-0.3.0-py3.9-linux-x86_64.egg/habitat_sim/simulator.py", line 131, in close
AttributeError: 'HabitatSim' object has no attribute '_initialized'
```

zhai-create, 2023-12-24:
I have completed the configuration of EquirectangularDepthSensor in the latest version of habitat, thank you.

closed 2023-12-24 by zhai-create