facebookresearch/habitat-lab #1702: Using .glb file in the simulation

alre5639, 2023-11-28:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: vx.x.x or master?

Habitat-Sim: vx.x.x or master?

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

aclegg3, 2023-11-28:
Hey @alre5639,

The dataset you are modifying here is actually a PointNavDataset which contains PointNavEpisodes. These are produced by the generator: https://github.com/facebookresearch/habitat-lab/tree/main/habitat-lab/habitat/datasets/pointnav. This generator is fed a configuration file which defines the assets used for the episodes (i.e. the SceneDataset). When you load the episode dataset the simulator will look for the assets which were used to generate those episodes.

TL;DR: You need to run the episode generator or get some HM3D PointNav episodes from elsewhere to load them in a lab environment.

Check the DATASET readme: https://github.com/facebookresearch/habitat-lab/blob/main/DATASETS.md for some pre-generated episodes if you don't want to generate your own.

alre5639, 2023-11-28:
Thank you for the quick response. I am actually not particularly interested in the point to point navigation aspects of this problem, I am really interested in unguided exploration in the downloaded meshes, this example just seemed to be quite close to what I was looking for. 

Is there another type rather than the `pointnav-v1` I could switch over to allow for exploration in the .glb files, or another example you could point me at for navigating within the .glb files? Thanks again.

jdiazram, 2024-10-23:
> Thank you for the quick response. I am actually not particularly interested in the point to point navigation aspects of this problem, I am really interested in unguided exploration in the downloaded [...]

Could you solve your problem? I'm looking for something similar, only exploration in HM3D.

THX

rainbow979, 2024-11-20:
I'm also trying to solve this problem. Recommend to refer to https://gist.github.com/erikwijmans/1c596183fb8de1e9afd80088b6b5c115