facebookresearch/habitat-lab #1704: Warning in SocialNav (Instantiating render asset ... with incompatible light setup, instance will not be correctly lit.)

qwerty863, 2023-11-29:
Thank you for the updated README for SocialNav!

Thanks to the updated README, I can successfully run training a SocialNav agent, but I see many warnings like below.
`[Warning]:[Assets] ResourceManager.cpp(945)::createRenderAssetInstance : Instantiating render asset 'data/robots/hab_spot_arm/urdf/../meshesColored/arm0.link_el0.glb' with incompatible light setup, instance will not be correctly lit.`

Have you observed this warning before? Would it affect reproducing the result?

jimmytyyang, 2023-12-03:
Hi @qwerty863! Yes, we have observed this before, and this should be okay. It would not affect the results.

qwerty863, 2023-12-04:
Hi @jimmytyyang! Thank you for confirming this!

closed 2023-12-04 by qwerty863