facebookresearch/habitat-lab #1708: Starting Point from pointnav quickstart .json not located in .glb

alre5639, 2023-12-01:
After unzipping the `habitat-test-scenes/v1/test/test.json.gz` episode 0 has a start position of `[-1.2676633596420288, 0.2047852873802185, 12.595427513122559]` which from running the sim seems to be one of the corners of the large room. After using open3d to review the full `skokloster-castle.glb` file to view and adding coordinate frames the (0,0,0) of the scene is located outside the scene (which is fine assuming a transformation to the "start_position" exists).  apply the translation from "start position" would result in a large z translation, which cant be right considering floor looks to be ~ 0 in this sim. Additionally given that "goal position" has a large z translation it seems the coordinate frame is rotated. However the start_rotation doesn't seem to provide a transform from the world frame (what i called (0,0,0) above) and the robot frame. In fact the (if the rotation is in quaterneons) it is simply ~45 degree rotation about the y axis. Given this building a homogenous transform matrix from the given translation and rotation information doesnt provide the correct starting location of the robot. 

How do I use the starting position from the test.json.gz file to find the location of the agent with respect to the .glb vertices coordinates?

test.json:
```
{"episodes": [{"episode_id": "0", "scene_id": "data/scene_datasets/habitat-test-scenes/skokloster-castle.glb", "start_position": [-1.2676633596420288, 0.2047852873802185, 12.595427513122559], "start_rotation": [0, 0.4536385088584658, 0, 0.8911857849408661], "info": {"geodesic_distance": 6.335183143615723, "difficulty": "easy"}, "goals": [{"position": [2.2896811962127686, 0.11950381100177765, 16.97636604309082], "radius": null}], "shortest_paths": null, "start_room": null}, {"episode_id": "1", "scene_id": "data/scene_datasets/habitat-test-scenes/skokloster-castle.glb", "start_position": [5.658442974090576, 0.00991860032081604, 19.127395629882812], "start_rotation": [0, 0.918212155009143, 0, -0.3960889273805385], "info": {"geodesic_distance": 11.762749671936035, "difficulty": "medium"}, "goals": [{"position": [6.113696575164795, 0.20991861820220947, 7.93007230758667], "radius": null}], "shortest_paths": null, "start_room": null}, 

...

``

![world_frame_castle](https://github.com/facebookresearch/habitat-lab/assets/88396371/17bd3fb6-3fdf-48ae-aa59-d4d29d789407)

aclegg3, 2023-12-04:
Hey @alre5639 

That starting coordinate is in Habitat's global coordinate system (Y up, Z forward) which is different from some other standard coordinate systems. In this image I assume rgb of the frame corresponds to xyz axes, so the "up" direction here is Z. This means the default behavior of your visualizer (open3d?) is to load this asset in a different global coordinate system. The difference would be a 90 degree rotation about X axis in this case to align the green axis (Y) with scene "up".

alre5639, 2023-12-04:
yep, I have confirmed this behavior and am able to locate my pose in the scene. @aclegg3  thank you for the support!

closed 2023-12-04 by alre5639