facebookresearch/habitat-lab #1709: error about Key 'count_coll_pen' not in 'SocialNavReward'

Li-ChangHao, 2023-12-03:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.3.0

Habitat-Sim: v0.3.0

tag：stable

## 🐛 Bug

When I run the code according to the readme, an error occurs：Key 'count_coll_pen' not in 'SocialNavReward'。

The following is the main error message：

[image]

jimmytyyang, 2023-12-03:
Hi @Li-ChangHao! Could you please try to `git pull origin main` to make sure you are using the latest habitat-lab `main`? The stable version might not work. In addition, would you please also paste your latest commit hash for habitat-lab? Thank you!

Li-ChangHao, 2023-12-04:
latest commit hash：afe4058a7f8aa5ab71a133575cdaa79f0308af6a

Li-ChangHao, 2023-12-12:
it works, many thanks

closed 2023-12-12 by Li-ChangHao