facebookresearch/habitat-lab #1715: How to generate SMPLX urdf data?

zhuhaozh, 2023-12-13:
Does anyone know how to generate the urdf dataset of a smplx? 

Habitat uses urdf to load the agents, and I want to add more human characters to habitat. 
I found the urdf files in `habitat_humanoid` dataset, but I dont know how to generate a new one. Is there any scripts can be used to generate urdf from the smplx data?

xavierpuigf, 2023-12-14:
Those are still not available. We are working on uploading the script, but it needs some cleanup. It will be up no later than Monday.

xavierpuigf, 2023-12-18:
Here is a script and README for generating a URDF dataset of SMPL-X body models. This branch should be compatible with main, besides this generation script. Let me know if you run into issues.

https://github.com/facebookresearch/habitat-lab/tree/addon_humanoid/habitat-lab/habitat/articulated_agents/humanoids#generating-new-shapes

zhuhaozh, 2023-12-19:
> Here is a script and README for generating a URDF dataset of SMPL-X body models. This branch should be compatible with main, besides this generation script. Let me know if you run into issues. [...]

Thanks so much for your great help!

zhuhaozh, 2023-12-19:
Hi @xavierpuigf , 
Thanks for your work. I found several bugs, and solved part of them, but it still doesn't work. I ran the script outside of the habitat-lab repo, but that may not be the problem?

I fork the repo and pushed a new commit, you can see the comparsion here: [link](https://github.com/zhuhaozh/habitat-lab/commit/2ebd372793e2d710176da6d10c2960cde7ef8765). I use "XXX" comment to mark the codes which I confused. 

Here are the codes that does not work:
1. Texture: 
- the user provided texture hasn't been used ([code](https://github.com/zhuhaozh/habitat-lab/blob/2ebd372793e2d710176da6d10c2960cde7ef8765/scripts/export_smplx_bodies.py#L518))
- the random texture files are not found ([code](https://github.com/zhuhaozh/habitat-lab/blob/2ebd372793e2d710176da6d10c2960cde7ef8765/scripts/export_smplx_bodies.py#L521))

2. Export URDF
- raise an error "AttributeError: Calling operator "bpy.ops.export_armature.urdf" error, could not be found", should I install some blender addons? ([code](https://github.com/zhuhaozh/habitat-lab/blob/2ebd372793e2d710176da6d10c2960cde7ef8765/scripts/export_smplx_bodies.py#L605))
- the function 'export_urdf' (I renamed to 'export_urdf_file') never be used, and I dont know where to find the 'settings' ([code](https://github.com/zhuhaozh/habitat-lab/blob/2ebd372793e2d710176da6d10c2960cde7ef8765/scripts/export_smplx_bodies.py#L358))

xavierpuigf, 2023-12-19:
Thanks for your feedback. Could you try to run the script again, after pulling? These should be resolved. I changed the code to load only the textures provided in the SMPL-X addon. We will be providing more textures soon.

zhuhaozh, 2023-12-20:
Thanks! The new script works fine after the bugs fixed: 
1.  body_file -> body-file, to align with the readme: [here](https://github.com/zhuhaozh/habitat-lab/commit/2ebd372793e2d710176da6d10c2960cde7ef8765#diff-610f0bd1c5ebf434162dc817d7b672d36b495bfa6694e2ff92d8f26bc6ebf14aR556)
2. load json, and warp it to a list: [here](https://github.com/zhuhaozh/habitat-lab/commit/2ebd372793e2d710176da6d10c2960cde7ef8765#diff-610f0bd1c5ebf434162dc817d7b672d36b495bfa6694e2ff92d8f26bc6ebf14aR481)
3. add a new line to set gender: [here](https://github.com/zhuhaozh/habitat-lab/commit/2ebd372793e2d710176da6d10c2960cde7ef8765#diff-610f0bd1c5ebf434162dc817d7b672d36b495bfa6694e2ff92d8f26bc6ebf14aR509)
4. break, if ind == 10. it works, but I'm not sure this modification is correct: [here](https://github.com/zhuhaozh/habitat-lab/commit/2ebd372793e2d710176da6d10c2960cde7ef8765#diff-610f0bd1c5ebf434162dc817d7b672d36b495bfa6694e2ff92d8f26bc6ebf14aR556)
5. remove the hardcode path: [here](https://github.com/facebookresearch/habitat-lab/blob/addon_humanoid/scripts/export_smplx_bodies.py#L616)

xavierpuigf, 2023-12-20:
Thanks for your help! I had not pushed the latest changes, I pushed a new commit based on your fixes and what I had before. Hopefully it should work now. Note that you can still set any texture for the SMPL-X avatars, but you need to place it inside the addon folder (where the other default textures are).

closed 2023-12-20 by xavierpuigf

ETwilight, 2024-07-19:
> Here is a script and README for generating a URDF dataset of SMPL-X body models. This branch should be compatible with main, besides this generation script. Let me know if you run into issues. [...]

Dear researcher, this link seems not to be working anymore. Is there an alternative link? Thank you!

xavierpuigf, 2024-07-19:
Here is the updated link! 
https://github.com/facebookresearch/habitat-lab/tree/main/habitat-lab/habitat/articulated_agents/humanoids#generating-new-shapes

ETwilight, 2024-07-23:
Thank you for providing the new link! I have a question regarding the code:
In my understanding (please correct me if I am wrong) this is a SMPL .npz -> .urdf converter.
But the MotionConverterSMPLX class in humanoid_utils.py seems to require an URDF file.
I will greatly appreciate if you could explain a bit of what this class is working, and, if I am wrong in understanding, what is the correct way of converting SMPL  npz to URDF format. Thank you so much in advance!