facebookresearch/habitat-lab #1721: Social Navigation Baselines: Spot legs are not articulated/animated in rendering

jlidard, 2023-12-18:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: 0.3.0

Habitat-Sim: 0.3.0

## Docs and Tutorials
Did you read the docs? Yes

Did you check out the tutorials? Yes

## ❓ Questions and Help

jimmytyyang, 2023-12-19:
Hi @jlidard! I hope you are doing well! Yes, you are correct, currently in the code we teleport the robot's base based on linear and angular velocities output from the social nav policy, and the legs are **not** animated. We will add animation code for legs later. In addition, the warning is not related to the leg animation. It is expected, so you do not need to worry about this.

jimmytyyang, 2023-12-19:
Thank you for reporting this! :)

jlidard, 2023-12-20:
Gotcha, thanks for clarifying @jimmytyyang!

closed 2023-12-20 by jlidard

ruddradev, 2024-02-06:
For anyone looking for it, I found the working code present in the below merge request and was able to run the leg animation successfully - 

https://github.com/facebookresearch/habitat-lab/pull/1554

licongfei001, 2024-11-08:
Hi @jlidard! Hope you are doing well! I am trying the social navigation baseline recently. When I try to train it. ERROR: AssertionError: Object with handle '004_sugar_box_:0000' could not be found in the scene. Has the Episode been initialized? When I try to evaluate it. ERROR: AttributeError: 'RearrangeSim' object has no attribute '_initialized'. Do you encounter the same errors?