facebookresearch/habitat-lab #1728: Why does the semantic sensor output perspective differ from RGB and depth sensors?

BaiLing09, 2023-12-25:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: tags challenge-2022

Habitat-Sim: tags challenge-2022

Datasets: MP3D sence and its object navigation task episodes

## ❓ Questions and Help
Questions:  the semantic sensor output perspective differ from RGB and depth sensors. (I try to print the sensor state. This three kinds of sensor are same states)

Following is my code:
config_paths = "/Configs/Object_Nav_mp3D.yaml"  # 输入habitat-lab的配置文件路径
config_paths = "/configs/challenge_objectnav2022.local.rgbd.yaml"
config = get_config(config_paths)

hab_env = Env(config=config)
print(hab_env._sim.get_agent_state())

action = {1: 'MOVE_FORWARD', 2: 'TURN_LEFT', 3: 'TURN_RIGHT'}
ep_i = 0
while ep_i < 10:
    observations = hab_env.reset()
    start = 0
    while start < 15:
        obs = hab_env.step(action[random.choice([1, 2, 3])])
        sematic, depth = dis(obs['semantic'], obs['depth'])   # dis is a function only for creating Visual image
        cv.imshow("RGB", obs['rgb'])
        cv.imshow("Semantic", sematic)
        cv.imshow("depth", depth)
        cv.waitKey(500)
        start += 1

Following is my config yaml:
[image]

Following is the running result:
[image]

aclegg3, 2024-01-04:
Hey @BaiLing09 

Most likely this is an issue with the coordinate system configuration for the scene meshes. In MP3D, the semantic mesh is a separate asset instantiating a separate SceneGraph. It is possible that the way the scene is initialized (e.g. linking directly to the glb file instead of using the SceneDataset configs) results in the correct scene transformation not being applied to the semantic mesh, therefore the 90 degree rotation difference between the semantic and rgb images.

Double check that you followed these directions for using the MP3D scenes: https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md#matterport3d-mp3d-dataset

Lanssi, 2025-12-24:
@JahaoZhang I followed the solution of @Yuxin916 , and it worked. Please refer to [here](https://github.com/facebookresearch/habitat-lab/issues/2196#issuecomment-3689916709) more detail