facebookresearch/habitat-lab #1737: Social Navigation: Easiest Way to Get Bird's Eye and Map Views From Evaluation Pipeline

jlidard, 2024-01-05:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.3.0

Habitat-Sim: v0.3.0

## Docs and Tutorials
Did you read the docs? Yes

Did you check out the tutorials? Yes

## ❓ Questions and Help

cc: @jimmytyyang 

Hi Habitat team, many thanks for the great release and for providing example code for evaluation. I was wondering what would be the easiest method for visualizing bird's eye and map views in the social navigation pipeline. Some of the example videos have these perspectives visualized (screenshots attached), but the provided pipeline doesn't generate these by default. What would be the easiest approach for rendering these views along with the videos? Could I add a `TopDownMapMeasurementConfig` as shown in [https://aihabitat.org/docs/habitat-lab/habitat-lab-tdmap-viz.html](this turorial), or would you suggest a different approach? Additionally, how can a bird's eye view be extracted? 

Thanks for your reply! 

[image]
[image]

xavierpuigf, 2024-01-07:
Hi @jlidard, `TopDownMapMeasurementConfig` should get you an abstract top-down map, as in the tutorial. To generate the videos you point to, we did something slightly different. It is not on `main` yet but it should not be too difficult to implement.

The main code that implements this function can be found on this [branch](https://github.com/facebookresearch/habitat-lab/blob/SIRo_dev_learn_skills_paperfig_web/habitat-lab/habitat/articulated_agents/humanoids/kinematic_humanoid.py#L210).

As you can see, we edit the code that manages cameras so that when `link_id == -3`, the camera transformation is not attached to any particular link (as in [here](https://github.com/facebookresearch/habitat-lab/blob/SIRo_dev_learn_skills_paperfig_web/habitat-lab/habitat/articulated_agents/humanoids/kinematic_humanoid.py#L225)) and is therefore static in the scene, without following the agent. Once you have defined that static camera, you can define a new birds-eye view camera, like `top` [here](https://github.com/facebookresearch/habitat-lab/blob/SIRo_dev_learn_skills_paperfig_web/habitat-lab/habitat/articulated_agents/humanoids/kinematic_humanoid.py#L48) (you want to make sure the link id is -3). The last step is to add cameras that are static in the scene, and with a transform that covers the scene. We define the camera [here](https://github.com/facebookresearch/habitat-lab/commit/73c6553ea5bb2d1755ed8dbf56b9d867377b34d8#diff-84b90de9b8a169889b52b3256ca546b3ea4b81daecca8f115e483e49497e47b6R129). In this code it is `fourth_rgb_sensor`, but the name should start such that it matches the camera you defined before in `kinematic_humanoid.py`, so in this case `top_rgb_sensor`.

Hope this helps!

jlidard, 2024-01-29:
@xavierpuigf Got it, thanks for your reply!

closed 2024-01-29 by jlidard