facebookresearch/habitat-lab #1740: google colab for habitat-Sim tutorial not working

PhilipAmadasun, 2024-01-07:
Operating System: Linux (Ubuntu 20.04)
I am trying to follow the tutorials provided but the google-colab environment is not set properly, I am unable to import git or habitat sim. I provide the link to the video and the google colab here:
[Tutorial Video](https://www.youtube.com/watch?v=kunFMRJAu2U&t=129s)
[Google-colab](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/v0.1.6/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb)
[Link to Tutorials](https://aihabitat.org/tutorial/2020/)

aclegg3, 2024-01-16:
Note we have stopped supporting colab and converted tutorials to local Jupyter notebooks instead.

closed 2024-01-16 by aclegg3

PhilipAmadasun, 2024-01-17:
@aclegg3 Were can I find the Jupyter notebooks?

aclegg3, 2024-01-17:
@PhilipAmadasun in #1742 we moved relevant notebooks to https://github.com/facebookresearch/habitat-lab/tree/main/examples/tutorials/notebooks

PhilipAmadasun, 2024-01-20:
@aclegg3  Thanks for the effort you're putting in to get these tutorials working on notebook. Will new tutorial videos also be created to supplement the notebooks as well? It also seems to me( I could be wrong) that the available notebooks do not  go in order of the google colab tutorials. Perhaps you are not yet done with the converting all the scripts? 

**On another topic** I could raise another issue but I just thought I might ask on here: Is the current dockerfile on the main branch still valid?