facebookresearch/habitat-lab #1741: where can I find the dataset 'goat_openvocab'?

wzjscut, 2024-01-08:
I use the branch home-robot_goat_support. but I cannot find the datasets path setting in  habitat/habitat-lab/habitat-lab/habitat/config/habitat/dataset/goat/hm3d.yaml 
data_path: **data/datasets/goat_openvocab/hm3d/v0.1.2_fixed/{split}/{split}.json.gz**

rahil-bhowal007, 2025-03-04:
did you find goat_openvocab dataset

aminkashiri, 2025-03-12:
@ZijianWu1121 I've the same issue. The closest thing I could find is the dataset provided in [goat-bench](https://github.com/Ram81/goat-bench?tab=readme-ov-file) repo: [link](https://drive.google.com/file/d/1N0UbpXK3v7oTphC4LoDqlNeMHbrwkbPe/view?usp=sharing). I then renamed the folders to match the "data_path". Let me know if this helps.