facebookresearch/habitat-lab #1751: Help

ChrisWanghb, 2024-01-13:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: 0.3.0 

Habitat-Sim: 0.3.0

AttributeError: 'habitat_sim._ext.habitat_sim_bindings.ManagedBulle' object has no attribute 'creation_attributes'

umuiq, 2024-01-15:
i met the same problem in 'conda install habitat-sim withbullet -c conda-forge -c aihabitat'
and running on 'python examples/interactive_play.py --never-end'

Leesuh1, 2024-01-18:
I also got a same error after executing below.

HABITAT_SIM_LOG=warning MAGNUM_LOG=warning \
python examples/hitl/basic_viewer/basic_viewer.py habitat_hitl.episodes_filter='0 2 4 10:15 1000:4000:500'

Leesuh1, 2024-01-18:
Just download newest version for sim and lab both!

shuishenbushui, 2024-01-19:
"Note, for newer features added after the most recent release, you may need to install aihabitat-nightly. See Habitat-Sim's [installation instructions](https://github.com/facebookresearch/habitat-sim#installation) for more details."
"We also provide a [nightly conda build for the main branch](https://anaconda.org/aihabitat-nightly). However, this should only be used if you need a specific feature not yet in the latest release version. To get the nightly build of the latest main, simply swap -c aihabitat for -c aihabitat-nightly."
So I install aihabitat-nightly:
conda install habitat-sim withbullet -c conda-forge -c aihabitat-nightly
the problem is solved.