facebookresearch/habitat-lab #1762: dataset download error

Thaumaturge2020, 2024-01-16:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: master

Habitat-Sim: master

when I try to download the habitat_test_scenes I got this error:

stderr: 'fatal: unable to access 'https://huggingface.co/datasets/ai-habitat/habitat_test_scenes.git/': gnutls_handshake() failed: Error in the pull function.'

the cmd I used was :

python -m habitat_sim.utils.datasets_download --uids habitat_test_scenes --data-path data/

How could i solve this problem?

21stYouth, 2024-01-25:
i have the same err in download, have you solved it?

21stYouth, 2024-01-26:
Unfortunately，i still have the problem.

Here is the err (when i use python -m habitat_sim.utils.datasets_download --uids habitat_test_scenes --data-path data/):

git.exc.GitCommandError: Cmd('git') failed due to: exit code(128)
  cmdline: git fetch -v -- origin v1.6
  stderr: 'fatal: unable to access 'https://huggingface.co/datasets/ai-habitat/ReplicaCAD_dataset.git/': Failed to connect to huggingface.co port 443 after 133412 ms: Connection timed out'

I went to the website https://huggingface.co/datasets/ai-habitat/ReplicaCAD_dataset.git/, it showed 404 to me. Is that normal, or just i have this result?

xavierpuigf, 2024-01-31:
I am not sure of what could be cause the timing out issue. As for the link, the way you can see the dataset on the website is by removing the `.git` extension.

https://huggingface.co/datasets/ai-habitat/ReplicaCAD_dataset

Lr-2002, 2024-02-07:
Are you in China Mainland？ This was because of the network ,you need to the proxy to solve it.

21stYouth, 2024-02-10:
Thanks everyone, i finally found the solution of these problems.
May i repeat the key point of my err:
When i use `python -m habitat_sim.utils.datasets_download --uids habitat_test_scenes --data-path data/` or `python examples/example.py` ,the program will have err and said 

> stderr: 'fatal: unable to access 'https://huggingface.co/datasets/ai-habitat/ReplicaCAD_dataset.git/'

So here are my solutions:
1. Git over SSH in Huggingface: You can see details in [https://huggingface.co/docs/hub/security-git-ssh](url) 
2. Find the git clone command in the err report, like 

> git clone --depth 1 --branch v1.6 https://huggingface.co/datasets/ai-habitat/ReplicaCAD_dataset.git /home/bingliangli/habitat-lab/data/versioned_data/replica_cad_dataset

3. Change the `https://huggingface.co/` to `git@hf.co:`. So the command will be like 

> git clone --depth 1 --branch v1.6 git@hf.co:datasets/ai-habitat/ReplicaCAD_dataset.git /home/bingliangli/habitat-lab/data/versioned_data/replica_cad_dataset

4. Use this command to download the datasets again and again when having the err, until all the datasets  are downloaded

21stYouth, 2024-02-10:
By the way, the problem of 

> subprocess.CalledProcessError: Command '['git', 'clone', '--depth', '1', '--branch', 'v1.6', 'https://huggingface.co/datasets/ai-habitat/ReplicaCAD_dataset.git', '/home/bingliangli/habitat-lab/data/versioned_data/replica_cad_dataset']' returned non-zero exit status 128.

maybe can be solved by this way as well

aclegg3, 2024-02-12:
@21stYouth 

HF has deprecated use of username/password for authentication. By using `git@hf.com` you specify use of the SSH authentication channel.
Our downloader [does this for authenticated datasets](https://github.com/facebookresearch/habitat-sim/blob/a29fe273bdecc6001b58e46c43321c69c45006ac/src_python/habitat_sim/utils/datasets_download.py#L578). I'm a bit surprised to see this is ever necessary for non-authenticated datasets.

juceyj, 2024-06-11:
@21stYouth Thanks for sharing your solution! 

After changing the command to:
`git clone --depth 1 --branch main git@hf.co:datasets/ai-habitat/hab3_episodes.git xxx/data/versioned_data/hab3-episodes`

There was a new error: 
`Error downloading object: checkpoint/social_nav_latest.pth (f48dd36): Smudge error: Error downloading checkpoint/social_nav_latest.pth (f48dd36d2cf8b41f810e3e754db6aaa0ca3c815ffdad08ab8d76473fa1f4d797): batch response: Post "https://huggingface.co/datasets/ai-habitat/hab3_episodes.git/info/lfs/objects/batch": read tcp 10.4.0.89:55310->54.230.71.56:443: read: connection reset by peer`

How could I solve that?

juceyj, 2024-06-11:
> @21stYouth Thanks for sharing your solution!
>
> After changing the command to: `git clone --depth 1 --branch main git@hf.co:datasets/ai-habitat/hab3_episodes.git [...]

Hey, the problem `connection reset by peer` has been solved by downloading dataset with VPN. It turns out people in China cannot download data in Huggingface, so I just download the dataset in a local device which has access to Huggingface and then upload it to the remote device.

Sellaris, 2024-12-24:
> @21stYouth Thanks for sharing your solution!
>
> After changing the command to: `git clone --depth 1 --branch main git@hf.co:datasets/ai-habitat/hab3_episodes.git [...]

Did u solve it now???I Meet the same problem.    :(