facebookresearch/habitat-lab #1804: `get_camera_transform` from `habitat.tasks.rearrange.utils` uses inconsistent getting method

zephirefaith, 2024-02-13:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.3.0 or master?

Habitat-Sim: v0.3.0 or master?

Habitat is under active development, and we advise users to restrict themselves to stable releases of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases). The bug you are about to report may already be fixed in the latest version.

Master branch contains 'bleeding edge' code, but we do appreciate bug reports for it!

## 🐛 Bug

The cameras attached to articulated agents are initialized using ArticulatedAgentCameraParams class [here](https://github.com/facebookresearch/habitat-lab/blob/main/habitat-lab/habitat/articulated_agents/mobile_manipulator.py#L19). The position for this sensor is set using the logic [here](https://github.com/facebookresearch/habitat-lab/blob/d4df9cc76cd3e5d66b24fbd92d004984a203196d/habitat-lab/habitat/articulated_agents/manipulator.py#L168-L183). However when `habitat.tasks.rearrange.utils.get_camera_transform` function is used to get the absolute transformation of this camera, the getter applies inconsistent logic taking only the translation component of camera's view-matrix. This leads to a camera-pose with right translation but wrong rotation wrt world frame.

## Steps to Reproduce

Steps to reproduce the behavior:

1. Initialize environment with Spot in it
2. Access camera pose using `get_camera_transform`
3. Access camera pose using `sim.agents[0]._sensors[<sensor-name>].render_camera.camera_matrix.inverted()`
4. Output of 2 != output of 3

See PCD built with 2 versus with 3 below.
[image]
[image]

SergioArnaud, 2024-03-09:
🤯

aclegg3, 2024-03-11:
Yeah, this is a pain. @SergioArnaud happy to review a fix if you want to tackle this. :smiley:

zephirefaith, 2024-03-11:
Sorry this made you stumble @SergioArnaud! @aclegg3 given this bug has a greater damage than population of 1, I can prioritize this and push a fix today.

zephirefaith, 2024-03-14:
Fixed in the PR linked above. Tested with zf/obs-wm-updates branch of habitat-llm. See attached videos of point-cloud from `get_camera_transform` method and `sim.agents[0]._sensors[<sensor-name>].render_camera.camera_matrix.inverted()` method. Camera used is `articulated_agent_jaw_rgb` and `articulate_agent_jaw_depth` introduced in #1856 .

https://github.com/facebookresearch/habitat-lab/assets/9580531/81ef2577-5060-4d80-b568-bbc86ebb6da0

https://github.com/facebookresearch/habitat-lab/assets/9580531/1fdc7095-d8fa-4e1b-a6ec-cdf873622b5b