facebookresearch/habitat-lab #1826: Clarification on the Purpose of Trainer and Agent

QixuanSun, 2024-03-01:
Hello,

I am working in ML, but I am new to Habitat. I'm slightly confused about the purpose of some files: hierarchical_trainer.py, nonlearning_agents.py, and robo_vln_trainer.py.

From my understanding, these files import functions from Habitat, which I expect are used for training purposes, given their filenames include "_trainer". And "agent" is for evaluation. However, the README mentions an offline data buffer, suggesting that the trainers extract data from this buffer, without live interaction with the Habitat environment. This has led to some confusion.

Could you please clarify the following:

The specific roles and functionalities of hierarchical_trainer.py, nonlearning_agents.py, and robo_vln_trainer.py.
Whether my understanding that the trainers operate by extracting data from an offline buffer, thus not requiring Habitat for real-time simulation, is correct.
I might be missing some context or misunderstanding the architecture, so the suggestions you give would help me understand the code.

Thank you very much for your time and assistance.

Josh

QixuanSun, 2024-03-01:
Sorry, I posted it in the wrong repo.

My issue is the failure of installation.
I was installing from https://github.com/jacobkrantz/VLN-CE using the code
git clone --branch v0.1.7 git@github.com:facebookresearch/habitat-lab.git
cd habitat-lab
# installs both habitat and habitat_baselines
python -m pip install -r requirements.txt
python -m pip install -r habitat_baselines/rl/requirements.txt
python -m pip install -r habitat_baselines/rl/ddppo/requirements.txt
python setup.py develop --all

However, I found this error below:
.....
01bfae789fa566a2be1f385d1/msgpack-1.0.7.tar.gz#sha256=572efc93db7a4d27e404501975ca6d2d9775705c2d922390d878fcf768d92c87
Best match: msgpack 1.0.7
Processing msgpack-1.0.7.tar.gz
Writing /tmp/easy_install-s3pm6ljt/msgpack-1.0.7/setup.cfg
Running msgpack-1.0.7/setup.py -q bdist_egg --dist-dir /tmp/easy_install-s3pm6ljt/msgpack-1.0.7/egg-dist-tmp-9o2qa8p4
warning: no files found matching '*.c' under directory 'msgpack'
cc1plus: warning: command line option '-Wstrict-prototypes' is valid for C/ObjC but not for C++
zip_safe flag not set; analyzing archive contents...
msgpack.__pycache__._cmsgpack.cpython-36: module references __file__
creating /home/q84sun/miniconda3/envs/vlnce/lib/python3.6/site-packages/UNKNOWN-0.0.0-py3.6-linux-x86_64.egg
Extracting UNKNOWN-0.0.0-py3.6-linux-x86_64.egg to /home/q84sun/miniconda3/envs/vlnce/lib/python3.6/site-packages
Adding UNKNOWN 0.0.0 to easy-install.pth file

Installed /home/q84sun/miniconda3/envs/vlnce/lib/python3.6/site-packages/UNKNOWN-0.0.0-py3.6-linux-x86_64.egg
error: The 'msgpack' distribution was not found and is required by webdataset

zhanglixuan0720, 2024-08-29:
same error