facebookresearch/habitat-lab #1836: Unable to load semantic annotation for Scene

edwardjjj, 2024-03-06:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.3.0

Habitat-Sim: v0.3.0

## 🐛 Bug

I'm creating an env of scene "00680-YmWinf3mhb5" but get the following warning message:
```
ResourceManager.cpp(369)::loadSemanticSceneDescriptor : SSD File Naming Issue! Neither SceneInstanceAttributes-provided name : `../data/scene_datasets/hm3d/train/00680-YmWinf3mhb5/YmWinf3mhb5.basis.scn` nor constructed filename : `../data/scene_datasets/hm3d/train/00680-YmWinf3mhb5/info_semantic.json` exist on disk.
```
Consequently, am getting empty list when calling the env.sim.semantic_annotations()

## Steps to Reproduce

Steps to reproduce the behavior:

### sim config
```yaml
type: Sim-v0
forward_step_size: 0.25
turn_angle: 30
create_renderer: false
requires_textures: true
auto_sleep: false
step_physics: true
concur_render: false
needs_markers: true
update_articulated_agent: true
scene: ../data/scene_datasets/hm3d/train/00680-YmWinf3mhb5/YmWinf3mhb5.basis.glb
scene_dataset: [./data/scene_datasets/hm3d/hm3d_annotated_basis.scene_dataset_config.json](https://file+.vscode-resource.vscode-cdn.net/home/edward/projects/habitat/src/data/scene_datasets/hm3d/hm3d_annotated_basis.scene_dataset_config.json)
additional_object_paths: []
seed: ${habitat.seed}
default_agent_id: 0
debug_render: false
debug_render_articulated_agent: false
kinematic_mode: false
should_setup_semantic_ids: true
debug_render_goal: true
robot_joint_start_noise: 0.0
ctrl_freq: 120.0
ac_freq_ratio: 4
load_objs: true
hold_thresh: 0.15
grasp_impulse: 10000.0
agents:
  main_agent:
    height: 0.88
    radius: 0.18
    max_climb: 0.2
    max_slope: 45.0
    grasp_managers: 1
    sim_sensors:
      rgb_sensor:
        type: HabitatSimRGBSensor
        height: 480
        width: 640
        position:
        - 0.0
        - 0.88
        - 0.0
        orientation:
        - 0.0
        - 0.0
        - 0.0
        hfov: 79
        sensor_subtype: PINHOLE
        noise_model: None
        noise_model_kwargs: {}
      depth_sensor:
        type: HabitatSimDepthSensor
        height: 480
        width: 640
        position:
        - 0.0
        - 0.88
        - 0.0
        orientation:
        - 0.0
        - 0.0
        - 0.0
        hfov: 79
        sensor_subtype: PINHOLE
        noise_model: None
        noise_model_kwargs: {}
        min_depth: 0.5
        max_depth: 5.0
        normalize_depth: true
      semantic_sensor:
        type: HabitatSimSemanticSensor
        height: 480
        width: 640
        position:
        - 0.0
        - 0.88
        - 0.0
        orientation:
        - 0.0
        - 0.0
        - 0.0
        hfov: 79
        sensor_subtype: PINHOLE
        noise_model: None
        noise_model_kwargs: {}
    is_set_start_state: false
    start_position:
    - 0.0
    - 0.0
    - 0.0
    start_rotation:
    - 0.0
    - 0.0
    - 0.0
    - 1.0
    joint_start_noise: 0.1
    joint_start_override: null
    articulated_agent_urdf: null
    articulated_agent_type: null
    ik_arm_urdf: null
    motion_data_path: ''
agents_order:
- main_agent
default_agent_navmesh: true
navmesh_include_static_objects: true
habitat_sim_v0:
  gpu_device_id: 0
  gpu_gpu: false
  allow_sliding: false
  frustum_culling: true
  enable_physics: false
  enable_hbao: false
  physics_config_file: [./data/default.physics_config.json](https://file+.vscode-resource.vscode-cdn.net/home/edward/projects/habitat/src/data/default.physics_config.json)
  leave_context_with_background_renderer: false
  enable_gfx_replay_save: false
ep_info: null
object_ids_start: 100
renderer:
  enable_batch_renderer: false
  composite_files: null
  classic_replay_renderer: false

```
### Build sim
```python
from habitat.sims import make_sim
sim = make_sim(id_sim=config.type, config=config)
```

## Expected behavior

I expect the simulation to properly load the semantic informations

edwardjjj, 2024-03-11:
# An Update to the cause. 
It seems that the scene_dataset field in the simulator config will be overwrite by the 
episode's `scene_dataset_config` field.
When I run debugger and step into the creation of env, I find the `scene_dataset_config` field is wrong. Is there anyway to change it?

[image]

aclegg3, 2024-03-11:
Yes, this is true. The episodes will define a dataset config based on the assets used to generate the episode. You could overwrite the dataset field within the episode configs (using python JSON library). Otherwise you're best off generating new episodes for the dataset you are using to be sure that the agent and object placements referenced within are valid for your assets.

Jaraxxus-Me, 2024-03-12:
Hi, a follow-up on this, I'm using habitat 0.2.3 (habitat-challenge 2023). I tried to enable semantic sensor in the Env, but I can only get all zero semantic map.
I'm using this config file:
```
defaults:
  - /habitat: habitat_config_base
  - /habitat/task: objectnav_v2_challenge
  - /habitat/simulator/agents@habitat.simulator.agents.main_agent: rgbds_agent
  - /habitat/dataset/objectnav: hm3d_v2_valmini
  - _self_

habitat:
  environment:
    max_episode_seconds: 500

  simulator:
    action_space_config: "velocitycontrol"
    agents:
      main_agent:
        sim_sensors:
          rgb_sensor:
            width: 480
            height: 640
            hfov: 42
            position: [0, 1.31, 0]
          semantic_sensor:
            type: HabitatSimSemanticSensor
            width: 480
            height: 640
            hfov: 42
            position: [0, 1.31, 0]
          depth_sensor:
            width: 480
            height: 640
            hfov: 42
            min_depth: 0.5
            max_depth: 5.0
            position: [0, 1.31, 0]
        height: 1.41
        radius: 0.17
        max_climb: 0.1
        cell_height: 0.05
    habitat_sim_v0:
      gpu_device_id: 0
      allow_sliding: False
 ```
 
 I used this code to generate the Env:
 ```
    overrides = []
    if args.action_space is not None:
        overrides.append("habitat/task/actions=" + args.action_space)
    env_config = get_config(benchmark_config_path, overrides)
    hab_env = Env(config=env_config)
```
I noticed a weird thing in the Env object, why is the "semantic" observation space having a value bound of 0~-1?
When I independently initialize a simulator using the same episode scene, I won't have this problem.
Could you provide some guidance on this? Thanks!

edwardjjj, 2024-03-15:
I have a follow up question about semantic annotations. In this [tutorial](https://aihabitat.org/docs/habitat-lab/habitat-sim-demo.html) I see that you can print hierarchical semantic information about the levels, regions and objects. 

However in HM3D dataset I'm using, I only get object level information. when accessing `semantic_scene.levels`, I get `[]` back. It might be because this error message:
```
[Metadata] AttributesManagerBase.h(398)::buildAttrSrcPathsFromJSONAndLoad : <Scene Instance> : No Glob path result found for `../data/scene_datasets/hm3d/train/00680-YmWinf3mhb5/*.basis.scene_instance.json` so unable to load templates from that path.
```
So where can I find these `basis.scene_instances.json` files?

edwardjjj, 2024-03-28:
I found the answer to the follow up question at [Habitat-Sim#2346](https://github.com/facebookresearch/habitat-sim/issues/2346)

closed 2024-03-28 by edwardjjj