facebookresearch/habitat-lab #1854: Add a new robot into one of the habitats scenes

EftePA, 2024-03-11:
How can I add a new robot into one of the habitats scenes? I have the URDF of the robot, but I don't know where I have to add it. And next step would be to define a controller for it, which that also I don't know.
I could just find this [answer](https://github.com/facebookresearch/habitat-sim/issues/2274#issuecomment-1971919946) but it was not clear for me what exactly I have to do, does anyone have any experiences in it? Thank you in advance.

xavierpuigf, 2024-04-02:
This tutorial should give a good overview. Check out MultiAgent section
https://github.com/facebookresearch/habitat-lab/blob/main/examples/tutorials/articulated_agents_tutorial.ipynb

mannytheshizzle, 2024-09-05:
@xavierpuigf is there a way to actually visualize the scene created in the articulated_agents tutorial in real time (i.e., using the habitat viewer like the HITL Apps do) instead of having to save videos. The viewer.py script only allows you to load a specific dataset and scene but how do i load my custom habitat environment config with the articulated robot in the scene. The interactive_play script still hasn't been resolved to best of my knowledge so it is unusable.

EftePA, 2024-09-25:
@xavierpuigf sorry for the long pause.  Now I'm back on this work and when I run the [articulated_agents_tutorial.ipynb](https://github.com/facebookresearch/habitat-lab/blob/main/examples/tutorials/articulated_agents_tutorial.ipynb), I get the error:
```
2(9) : warning C7547: extension GL_NV_geometry_shader_passthrough not supported in profile gp5gp
Internal error: assembly compile error for geometry shader at offset 95:
-- error message --
line 4, column 38:  error: program option type not supported on this hardware
-- internal assembly text --
```
Meanwhile, I can run benchmarks like play_spot and close_fridge  with no error.

So the question is, what does exactly this error mean? What is missing on my hardware? 
On the other hand, my final goal is to add Pepper robot to one of the available scenes, in that case, do I need to solve this error?