facebookresearch/habitat-lab #1863: Collision handling between multiple humanoid agents

Zeying-Gong, 2024-03-16:
## Habitat-Lab and Habitat-Sim versions

Habitat-Lab: v0.3.0

Habitat-Sim: v0.3.0

## Docs and Tutorials

Did you read the docs? https://aihabitat.org/docs/habitat-lab/ Yes.

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/ Yes.

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

I refer to #1696 and define a new humanoid agent (called agent_2), which has the same config as agent_1. However, the current configuration does not handle collisions between humanoid agents effectively, resulting in models overlapping or penetrating each other. I am interested in understanding if collisions between humanoid agents can be treated similarly to collisions with obstacles.

[image]

Furthermore, I have noticed the AreAgentsWithinThreshold sensor, which is supposed to indicate if agents are close to each other and on the verge of colliding, prompting them to replan. I am curious about the possibility of implementing collision detection for humanoid agents and automatically replanning their path to the original destination. Can this be achieved easily?

Thank you in advance!

xavierpuigf, 2024-04-01:
Hi! You are right, the current configuration does not handle physic forces due to collisions. However you can detect collisions between agents and apply any logic you want when that event happens. Happy to help on that. The reason why collisions do not work is that humanoids are kinematically simulated: they are updated via joint angles, and not dynamic forces. You can turn dynamic forces on but then the humanoids will collapse in the floor. Note that while humanoids are not affected by forces, they can affect other objects by forces. If you set some object to be dynamic and the human collides with it, the object will move.

> Furthermore, I have noticed the AreAgentsWithinThreshold sensor, which is supposed to indicate if agents are on the verge of colliding, prompting them to replan. I am confused about how to replan their path to the original destination, can this be achieved easily?

The replanning happens at the high level, prompting agents to switch to navigate to a different object when they are within some threshold. What you probably want is a dynamic navigation to avoid the agents. We developed a version of this where we compute the path of the agent so that it reaches some position while avoiding the other agent. You can see an old implementation here:

https://github.com/facebookresearch/habitat-lab/blob/bc7f65d13b5bc2df5a85febe4175668b756d7457/habitat-lab/habitat/tasks/rearrange/actions/oracle_nav_action.py#L627

Zeying-Gong, 2024-04-03:
Thank you for your detailed answer! This support is quite helpful for us beginners to simulate the multi-agent environment in Habitat. I will try it later. 😊

closed 2024-04-03 by Zeying-Gong