facebookresearch/habitat-lab #1888: DISPLAY not detected. For headless systems, compile with --headless for EGL support

Columba0376, 2024-04-02:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.1.7

Habitat-Sim: v0.1.7

after I install Habitat-Sim: v0.1.7 with headless and run the test code I meet the error:
DISPLAY not detected. For headless systems, compile with --headless for EGL support.

Here is what I did while installing:
```shell
conda install -c aihabitat -c conda-forge habitat-sim=0.1.7 headless
git clone --branch v0.1.7 git@github.com:facebookresearch/habitat-lab.git
cd habitat-lab
python setup.py develop --all # install habitat and habitat_baselines
### test code:
python examples/example.py
###then the running stop at:
DISPLAY not detected. For headless systems, compile with --headless for EGL support.
``` 
How can I solve this problem?
Thank you in advanced!

xavierpuigf, 2024-04-03:
Can you point to where you found these install instructions? You may want to follow the order of arguments as shown in the install section:

```
conda install habitat-sim=0.1.7 headless -c conda-forge -c aihabitat
```

Columba0376, 2024-04-03:
> Can you point to where you found these install instructions? You may want to follow the order of arguments as shown in the install section:
> 
> ```
> conda install habitat-sim=0.1.7 headless -c conda-forge -c aihabitat
> ```

I found the install instructions from another project:https://github.com/MarSaKi/ETPNav

However, after trying this code, I still get the same error
```
conda install habitat-sim=0.1.7 headless -c conda-forge -c aihabitat
```

xavierpuigf, 2024-04-03:
Could you try installing a newer version of Habitat? It may be that older versions require older nvidia drivers as well. Are you installing this in a machine with GPUs?

Columba0376, 2024-04-11:
> Could you try installing a newer version of Habitat? It may be that older versions require older nvidia drivers as well. Are you installing this in a machine with GPUs?

Thank you. It seem it can run correctly using docker instead of using conda env

closed 2024-04-12 by xavierpuigf

chenguanqi, 2024-08-05:
@Columba0376 Hi, which docker image do you use to solve this problem?