facebookresearch/habitat-lab #1919: https://github.com/facebookresearch/habitat-api.git

dongxinfeng1, 2024-04-23:
*** Check failure stack trace: ***
Traceback (most recent call last):
  File "/home/z/Neural-SLAM/main.py", line 769, in <module>
    main()
  File "/home//Neural-SLAM/main.py", line 119, in main
    envs = make_vec_envs(args)
  File "/home/z/Neural-SLAM/env/__init__.py", line 7, in make_vec_envs
    envs = construct_envs(args)
  File "/home/z/Neural-SLAM/env/habitat/__init__.py", line 102, in construct_envs
    range(args.num_processes))
  File "/home/z/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 117, in __init__
    read_fn() for read_fn in self._connection_read_fns
  File "/home/z/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 117, in <listcomp>
    read_fn() for read_fn in self._connection_read_fns
  File "/home/z/anaconda3/envs/ns37/lib/python3.7/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
  File "/home/z/anaconda3/envs/ns37/lib/python3.7/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/home/z/anaconda3/envs/ns37/lib/python3.7/multiprocessing/connection.py", line 379, in _recv
    chunk = read(handle, remaining)
ConnectionResetError: [Errno 104] Connection reset by peer
Exception ignored in: <function VectorEnv.__del__ at 0x7f13532034d0>
Traceback (most recent call last):
  File "/home/z/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 487, in __del__
    self.close()
  File "/home/z/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 351, in close
    write_fn((CLOSE_COMMAND, None))
  File "/home/z/anaconda3/envs/ns37/lib/python3.7/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/home/z/anaconda3/envs/ns37/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/home/z/anaconda3/envs/ns37/lib/python3.7/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe

Process finished with exit code 1

licongfei001, 2024-11-06:
I encountered the same error. Have you had a solution?

hldbalabala, 2025-01-21:
I also encountered the same error. Have you had a solution?

licongfei001, 2025-01-21:
No, I have tried several times but it still didn’t work, so I just gave up.Best regardsCongfei Li