facebookresearch/habitat-lab #1920: Question of TopDown Map with HSSD dataset

wzcai99, 2024-04-23:
## ❓ Questions and Help
Hi, I am running ObjectNav benchmark with HSSD dataset. But when I wants to output the topdown map as one of the metrics, which is ok with the HM3D dataset, I got the error in /habitat-lab/habitat/tasks/nav/nav.py line 772
`assert int(sem_scene.objects[object_id].id.split("_")[-1]`. And when I check the line 769,
`sem_scene = self._sim.semantic_annotations()` I found sem_scene.levels = [], sem_scene.objects=[], sem_scene.regions=[]. 
To get the topdown map, I add the following lines in the config file:
`habitat_config.habitat.task.measurements.update(
        {
            "top_down_map": TopDownMapMeasurementConfig(
                map_padding=3,
                map_resolution=1024,
                draw_source=True,
                draw_border=True,
                draw_shortest_path=False,
                draw_view_points=True,
                draw_goal_positions=True,
                draw_goal_aabbs=True,
                fog_of_war=FogOfWarConfig(
                    draw=True,
                    visibility_dist=5.0,
                    fov=79,
                ),
            ),
            "collisions": CollisionsMeasurementConfig(),
        })`

Reventh-Sharma, 2024-05-29:
Same error for me, were you able to do something about it?

matinaghaei, 2024-08-27:
I was able to get around the issue by commenting out /habitat-lab/habitat/tasks/nav/nav.py line 859 `self._draw_goals_aabb(episode)`. However, this needs a more proper fix.