facebookresearch/habitat-lab #1921: Strange black patches in rendering observations in hm3d v1 scenes

verityw, 2024-04-24:
I noticed that there are strange black rectangular patches in observations from scenes in hm3d v1. This seems to be *different* from the black missing mesh rendering spots mentioned [here](https://github.com/facebookresearch/habitat-lab/issues/402) -- the patches are consistently rectangular, rather than irregularly shaped like in said issue. Additionally, the patches are differently sized and distributed when the resolution changes -- I've attached images of the same views in the same scene at two resolutions, and the higher resolution one has more small patches.

In the past, I've only used v2 (which only has the irregular mesh black patches, not the rectangular ones displayed below). Thus, I'm not sure if this is an issue with just hm3d v1.

[Here's a short Jupyter notebook that generated the below images.](https://drive.google.com/file/d/1oRESFums-_GYUEy3ZK34L4Oq1OxCWc7-/view?usp=sharing)

Any help would be greatly appreciated!

[image]
[image]

## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.3.1
Habitat-Sim: v0.3.1

aclegg3, 2024-04-24:
This is almost definitely a graphics driver issue. 
@0mdc FYI

This may be related to: https://github.com/facebookresearch/habitat-sim/issues/1987 or https://github.com/facebookresearch/habitat-sim/issues/2310 (if using A100 GPU).

You may find some debugging tips here useful: https://github.com/facebookresearch/habitat-lab/blob/main/TROUBLESHOOTING.md

verityw, 2024-04-24:
Aha, it is using an A100. Weird, I've never had this issue in the past before. Let me check out those links...

verityw, 2024-04-27:
reboot and nvidia driver reinstall worked btw :) thanks for the help!

closed 2024-04-27 by verityw

jdiazram, 2024-10-29:
Hi @verityw. I have been trying to use hm3d in Habitat-Lab without good results. Is it possible to share the code on how to use hm3d in Habitat-Lab? This is the link you shared, but it is broken (https://drive.google.com/file/d/1oRESFums-_GYUEy3ZK34L4Oq1OxCWc7-/view?usp=sharing)
Thanks