facebookresearch/habitat-lab #1993: Humanoids_tutorial.ipynb: No such file or directory: 'data/hab3_bench_assets/episode_datasets/small_large.json.gz'

Skevinci, 2024-06-26:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: 0.3.1

Habitat-Sim: 0.3.1

## 🐛 Bug

When running the init env code in the examples/tutorials/humanoids_tutorial.ipynb, it report error that:

FileNotFoundError                         Traceback (most recent call last)
Cell In[4], [line 6](vscode-notebook-cell:?execution_count=4&line=6)
      [1](vscode-notebook-cell:?execution_count=4&line=1) # Define the actions
      [3](vscode-notebook-cell:?execution_count=4&line=3) action_dict = {
      [4](vscode-notebook-cell:?execution_count=4&line=4)     "humanoid_joint_action": HumanoidJointActionConfig()
      [5](vscode-notebook-cell:?execution_count=4&line=5) }
----> [6](vscode-notebook-cell:?execution_count=4&line=6) env = init_rearrange_env(agent_dict, action_dict)

Cell In[2], [line 46](vscode-notebook-cell:?execution_count=2&line=46)
     [44](vscode-notebook-cell:?execution_count=2&line=44) hab_cfg = make_hab_cfg(agent_dict, action_dict)
     [45](vscode-notebook-cell:?execution_count=2&line=45) res_cfg = OmegaConf.create(hab_cfg)
---> [46](vscode-notebook-cell:?execution_count=2&line=46) return Env(res_cfg)

File ~/research/habitat-lab/habitat-lab/habitat/core/env.py:88, in Env.__init__(self, config, dataset)
     [86](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/core/env.py:86) self._dataset = dataset
     [87](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/core/env.py:87) if self._dataset is None and config.dataset.type:
---> [88](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/core/env.py:88)     self._dataset = make_dataset(
     [89](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/core/env.py:89)         id_dataset=config.dataset.type, config=config.dataset
     [90](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/core/env.py:90)     )
     [92](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/core/env.py:92) self._current_episode = None
     [93](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/core/env.py:93) self._episode_iterator = None

File ~/research/habitat-lab/habitat-lab/habitat/datasets/registration.py:22, in make_dataset(id_dataset, **kwargs)
     [19](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/registration.py:19) _dataset = registry.get_dataset(id_dataset)
     [20](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/registration.py:20) assert _dataset is not None, "Could not find dataset {}".format(id_dataset)
---> [22](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/registration.py:22) return _dataset(**kwargs)

File ~/research/habitat-lab/habitat-lab/habitat/datasets/rearrange/rearrange_dataset.py:74, in RearrangeDatasetV0.__init__(self, config)
     [70](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/rearrange/rearrange_dataset.py:70)     logger.info("Downloaded and extracted the data.")
     [72](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/rearrange/rearrange_dataset.py:72) check_and_gen_physics_config()
---> [74](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/rearrange/rearrange_dataset.py:74) super().__init__(config)

File ~/research/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py:116, in PointNavDatasetV1.__init__(self, config)
    [112](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py:112)     return
    [114](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py:114) datasetfile_path = config.data_path.format(split=config.split)
--> [116](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py:116) self._load_from_file(datasetfile_path, config.scenes_dir)
    [118](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py:118) # Read separate file for each scene
    [119](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py:119) dataset_dir = os.path.dirname(datasetfile_path)

File ~/research/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py:105, in PointNavDatasetV1._load_from_file(self, fname, scenes_dir)
    [103](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py:103)         self.from_binary(pickle.load(f), scenes_dir=scenes_dir)
    [104](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py:104) else:
--> [105](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py:105)     with gzip.open(fname, "rt") as f:
    [106](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/research/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py:106)         self.from_json(f.read(), scenes_dir=scenes_dir)

File ~/miniforge3/envs/habitat/lib/python3.9/gzip.py:58, in open(filename, mode, compresslevel, encoding, errors, newline)
     [56](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/miniforge3/envs/habitat/lib/python3.9/gzip.py:56) gz_mode = mode.replace("t", "")
     [57](https://vscode-remote+ssh-002dremote-002b141-002e212-002e78-002e253.vscode-resource.vscode-cdn.net/home/sikaili/research/habitat-lab/examples/tutorials/~/miniforge3/envs/habitat/lib/python3.9/gzip.py:57) if isinstance(filename, (str, bytes, os [... cut ...]
FileNotFoundError: [Errno 2] No such file or directory: 'data/hab3_bench_assets/episode_datasets/small_large.json.gz'

## Steps to Reproduce

Steps to reproduce the behavior:

1. Follow the installation in the README of habitat-lab
2. Follow the humanoids_tutorial.ipynb under example/ till "env = init_rearrange_env(agent_dict, action_dict)"

Please note that without a minimal working example to reproduce the bug, we may not be able to help you.

## Expected behavior

env is init and I can step forward to the tutorial

## Additional context

xavierpuigf, 2024-06-28:
You can find the dataset here: https://huggingface.co/datasets/ai-habitat/hab3_bench_assets/tree/main/episode_datasets
It can be downloaded using the command:

```
python -m habitat_sim.utils.datasets_download --uids hab3_bench_assets
```
Will update the tutorial to show that.

theutsavpanchal, 2024-11-25:
> You can find the dataset here: https://huggingface.co/datasets/ai-habitat/hab3_bench_assets/tree/main/episode_datasets It can be downloaded using the command:
> 
> ```
> python -m habitat_sim.utils.datasets_download --uids hab3_bench_assets
> ```
> 
> Will update the tutorial to show that.

I downloaded the hab3_bench_assets and it is still throwing the same error. 
`ValueError: Requested RearrangeDataset config paths 'data/hab3_bench_assets/episode_datasets/small_large.json.gz' or 'data/scene_datasets' are not downloaded locally. Aborting.`
Eventhough the small_large_json.gz is still there in the /data folder.

 
[image]

Roihn, 2025-06-12:
> I downloaded the hab3_bench_assets and it is still throwing the same error. `ValueError: Requested RearrangeDataset config paths 'data/hab3_bench_assets/episode_datasets/small_large.json.gz' or [...]

The same to me here. Wondering if you have figure out the solution. Thanks