facebookresearch/habitat-lab #1997: Question about missing scene rendering

zoeyliu1999, 2024-07-05:
Dear authors, thank you for your excellent work!

I am trying to render a scene with a specified motion file. I removed the following line from the code:

https://github.com/facebookresearch/habitat-lab/blob/01f33323692dfe573d3b4072a3e90c3d5f3c3a5c/test/test_humanoid.py#L166

As a result, the default scene should be data/scene_datasets/habitat-test-scenes/skokloster-castle.glb. I expected the humanoid to walk in the Skokloster Castle, but the scene is missing in the visualization video.

Here is one frame from the video:

[image]

Am I missing something that is causing the rendering to fail? Any suggestions or guidance would be greatly appreciated. Thank you!

xavierpuigf, 2024-07-05:
I think that the scene is still there, but it is in a different location. You can verify that the scene is created by running this before the test ends:

```
from habitat.sims.habitat_simulator.debug_visualizer import DebugVisualizer
import cv2
        
IM_W = 256
dbv = DebugVisualizer(sim, resolution=(IM_W, IM_W))
dbo = dbv.peek("scene")
img_scene = np.array(dbo.get_image())[:, :, [2,1,0]]
cv2.imwrite("res.png", img_scene)
         
```
You should get this, which is an overlap between the scene and the floor object we create.
[image]

You can get the location and bounds of the scene by running:

```
 sim.get_active_scene_graph().get_root_node().cumulative_bb
```

zoeyliu1999, 2024-07-06:
> I think that the scene is still there, but it is in a different location. You can verify that the scene is created by running this before the test ends:
> 
> ```
> from habitat.sims.habitat_simulator.debug_visualizer import DebugVisualizer
> import cv2
>         
> IM_W = 256
> dbv = DebugVisualizer(sim, resolution=(IM_W, IM_W))
> dbo = dbv.peek("scene")
> img_scene = np.array(dbo.get_image())[:, :, [2,1,0]]
> cv2.imwrite("res.png", img_scene)
>          
> ```
> 
> You should get this, which is an overlap between the scene and the floor object we create. <img alt="image" width="887" src="https://private-user-images.githubusercontent.com/4203263/346238068-dd30ae03-cb0f-49ec-94ec-f4d75402a77c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAyMzM5MjcsIm5iZiI6MTcyMDIzMzYyNywicGF0aCI6Ii80MjAzMjYzLzM0NjIzODA2OC1kZDMwYWUwMy1jYjBmLTQ5ZWMtOTRlYy1mNGQ3NTQwMmE3N2MucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MDZUMDI0MDI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NGViZGYzOWQ0ZGRhZGU2ZDNiMjliZTEwYTc1MWViYzJlOTJjMWFkNmZlMWY4YzAwMmE4ODQ2NjNiMWM2NmU1OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.ZohZG0iXOf0veyR8mB3Yx6rAPbB7ay_mWU-TUDjXrmk">
> 
> You can get the location and bounds of the scene by running:
> 
> ```
>  sim.get_active_scene_graph().get_root_node().cumulative_bb
> ```

You are right! After running the `DebugVisualizer`, I observed exactly the overlap between the scene and the floor object.

[image]

The location and bounds of the scene are: `Range({-9.75916, -0.390077, 0.973852}, {8.56903, 6.83441, 25.5983})`. Does this mean I should retarget the location of the humanoid?

xavierpuigf, 2024-07-08:
That's right! Retargeting should do

closed 2024-08-12 by xavierpuigf