facebookresearch/habitat-lab #2045: Instance Segmentation for Humanoid Agents

J0HNN7G, 2024-08-27:
## Habitat-Lab and Habitat-Sim versions

Habitat-Lab: 0.3.1
Habitat-Sim: 0.3.1

## ❓ Questions and Help

Can you get unique instance IDs for humanoid agents in semantic sensor observations?

In __RearrangeSim__, when I get observations from a semantic sensor with several humanoid agents in the camera view, they all have the same semantic ID.

wsakobe, 2024-10-27:
Hi, I’m experiencing the same issue. Have you found a solution? Thanks!

aclegg3, 2024-10-28:
Hey @wsakobe and @jonathanfrennert 

You are correct that in the current implementation, all humanoids are set to the same semantic id. This is something which should change down the line and there is a work around.

Currently, the semantic id is set in the `*.ao_config.json` file accompanying the humanoid assets. 
E.g. female_0.ao_config.json -
```{"render_asset": "female_0.glb", "debug_render_primitives": false, "semantic_id": 100}```

It is a bit hacky, but you can change this programmatically after the humanoid is loaded by changing the [SceneNode.semantic_id](https://aihabitat.org/docs/habitat-sim/habitat_sim.scene.SceneNode.html#semantic_id) property.
```
#assuming "humanoid" is the KinematicHumanoid object
for node in humanoid.sim_obj.visual_scene_nodes:
  node.semantic_id = <desired_semantic_id>
```

Note that any task sensors specifically created for detecting humans may be brittle with regard to this work-around. They may be looking for the specific semantic id "100". I'm not sure, so tread carefully. Hope this helps unblock you for now.

jonathanfrennert, 2024-10-28:
Thank you for the response!

For my use-case, I did not need to have the humans be articulated during simulation, so I ended up generating the humans in different poses beforehand and loading them as rigid objects.

aclegg3, 2024-10-28:
@jonathanfrennert yeah, that seems like a good solution then. Glad it worked out for you. :)

wsakobe, 2024-10-29:
@aclegg3 Thank you for your valuable and timely response; it has been a great help.

I tried modifying the corresponding semantic_id when initializing the humanoid, as shown in your code. However, my `humanoid.sim_obj.visual_scene_nodes` is empty, so I was unable to successfully change this value. Nevertheless, since each humanoid in my application corresponds to a different asset, I directly modified the `*.ao_config.json` file, which achieved the same effect.

wsakobe, 2024-10-29:
@aclegg3 Additionally, I noticed that in Habitat, when switching to the next episode using `env.reset()` after defining the Env, if the `scene_id` of the next episode does not match the current one, a significant increase in memory usage is observed after switching simulators. Is this behavior normal? How can this portion of memory be released?

aclegg3, 2024-10-29:
> when switching to the next episode using env.reset() after defining the Env, if the scene_id of the next episode does not match the current one, a significant increase in memory usage is observed

This is expected. Habitat employs a dynamic loading scheme so assets are not processed from disk or loaded onto the GPU until they are needed. This means that the first time you load a scene all the assets are processed which increase memory load on the system.

To release memory, you should call `Simulator.close(destroy=True)`. Alternatively you can use `with Simulator as sim:` syntax so the close happens automatically when the Simulator goes out of scope. Depends on your code and use case.

aclegg3, 2024-10-29:
Given the topic is drifting, I'll close this issue. Feel free open another with further questions.

closed 2024-10-29 by aclegg3