facebookresearch/habitat-lab #2063: Examples.py giving error!

MeltedHyperion07, 2024-09-02:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: master

Habitat-Sim: master

Habitat is under active development, and we advise users to restrict themselves to stable releases of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases). The bug you are about to report may already be fixed in the latest version.

Master branch contains 'bleeding edge' code, but we do appreciate bug reports for it!

## 🐛 Bug
I am running the code on my Ubuntu 22.04.4 workstation with Nvidia A5000 GPU with 24GB VRAM.
I am trying to replicate the steps in the readme of habitat lab website but when I try to execute the command "python examples/example.py" given in the "Non-interactive testing" part, it is throwing some error stating the following:  

## Steps to Reproduce

Steps to reproduce the behavior:
```
╭─    ~/habitat-lab   #v0.3.1 ?2 ▓▒░·······························································································································································································░▒▓ ✔  4s     habitat   14:23:27  
╰─ python examples/example.py
pybullet build time: Sep  2 2024 10:44:21
2024-09-02 14:23:40,719 Initializing dataset RearrangeDataset-v0
2024-09-02 14:23:53,859 initializing sim RearrangeSim-v0
[14:23:53:873274]:[Warning]:[Metadata] AttributesManagerBase.h(457)::buildAttrSrcPathsFromJSONAndLoad : <Articulated Object> : No Glob path result found for `data/hab_fetch_1.0/robots/fetch_no_base.urdf` so unable to load templates from that path.
[14:23:53:908563]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_00.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908597]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_01.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908615]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_02.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908630]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_03.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908645]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_04.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908660]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_05.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908677]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_06.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908693]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_07.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908711]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_08.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908725]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_09.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908743]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_10.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908760]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_11.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908776]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_12.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908792]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_13.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908807]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_14.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908822]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_15.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908836]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_16.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908853]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_17.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908867]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_18.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908882]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_19.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
[14:23:53:908897]:[Error]:[Metadata] SceneDatasetAttributesManager.cpp(304)::validateMap : `navmesh_instances` Value : `navmeshes/v3_sc4_staging_20.navmesh` not found on disk as absolute path or relative to `data/replica_cad`
Renderer: NVIDIA RTX A5000/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 535.183.01
Using optional features:
    GL_ARB_vertex_array_object
    GL_ARB_separate_shader_objects
    GL_ARB_robustness
    GL_ARB_texture_storage
    GL_ARB_texture_view
    GL_ARB_framebuffer_no_attachments
    GL_ARB_invalidate_subdata
    GL_ARB_texture_storage_multisample
    GL_ARB_multi_bind
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_texture_filter_anisotropic
    GL_KHR_debug
    GL_KHR_parallel_shader_compile
    GL_NV_depth_buffer_float
Using driver workarounds:
    no-forward-compatible-core-context
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
[... 1 lines cut ...]
MeshTools::compile(): ignoring Trade::MeshAttribute::TextureCoordinates 1 as its binding slot is already occupied by Trade::MeshAttribute::TextureCoordinates 0
MeshTools::compile(): ignoring Trade::MeshAttribute::TextureCoordinates 1 as its binding slot is already occupied by Trade::MeshAttribute::TextureCoordinates 0
MeshTools::compile(): ignoring Trade::MeshAttribute::TextureCoordinates 1 as its binding slot is already occupied by Trade::MeshAttribute::TextureCoordinates 0
MeshTools::compile(): ignoring Trade::MeshAttribute::TextureCoordinates 1 as its binding slot is already occupied by Trade::MeshAttribute::TextureCoordinates 0
[14:23:54:452079]:[Warning]:[Sim] Simulator.cpp(594)::instanceStageForSceneAttributes : The active scene does not contain semantic annotations : activeSemanticSceneID_ = 0
2024-09-02 14:23:56,539 Initializing task RearrangePickTask-v0
/home/administrator/miniconda3/envs/habitat/lib/python3.9/site-packages/gym/spaces/box.py:84: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(f"Box bound precision lowered by casting to {self.dtype}")
Environment creation successful
[... cut ...]
Bad inertia tensor properties, setting inertia to zero for link: l_gripper_finger_link
b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,467]:
Bad inertia tensor properties, setting inertia to zero for link: r_gripper_finger_link
b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,467]:
Bad inertia tensor properties, setting inertia to zero for link: l_gripper_finger_link
Agent acting inside environment.
Episode finished after 27 steps.

```

## Expected behavior

It should've run the example.

Claessens14, 2024-10-27:
I got a similar error, it is looking for sc4 files but I don't have them 
```
    apple-buffer-texture-unbind-on-buffer-modify
[07:01:29:166727]:[Metadata] MetadataMediator.cpp(297)::getSceneInstanceAttributesByName : Dataset : data/replica_cad/replicaCAD.scene_dataset_config.json has no preloaded SceneInstanceAttributes or StageAttributes named : data/replica_cad/configs/scenes/v3_sc4_staging_16.scene_instance.json so loading/creating a new StageAttributes with this name, and then creating a SceneInstanceAttributes with the same name that references this stage.
[07:01:29:166751]:[Metadata] AttributesManagerBase.h(352)::createFromJsonOrDefaultInternal : <Stage Template>: Proposing JSON name : data/replica_cad/configs/scenes/v3_sc4_staging_16.scene_instance.stage_config.json from original name : data/replica_cad/configs/scenes/v3_sc4_staging_16.scene_instance.json| This file does not exist.
[07:01:29:166843]:[Metadata] StageAttributesManager.cpp(86)::registerObjectFinalize : Render asset template handle : data/replica_cad/configs/scenes/v3_sc4_staging_16.scene_instance.json specified in stage template with handle : data/replica_cad/configs/scenes/v3_sc4_staging_16.scene_instance.json does not correspond to any existing file or primitive render asset.  Aborting.
[07:01:29:168581]:[Scene] SceneManager.h(25)::~SceneManager : Deconstructing SceneManager
[07:01:29:168609]:[Gfx] Renderer.cpp(72)::~Impl : Deconstructing Renderer
[07:01:29:168614]:[Gfx] WindowlessContext.h(17)::~WindowlessContext : Deconstructing WindowlessContext
Traceback (most recent call last):
  File "/Users/jacobclaessens/Desktop/code/habitat-2022/habitat-lab/dev/run.py", line 72, in <module>
    main()
  File "/Users/jacobclaessens/Desktop/code/habitat-2022/habitat-lab/dev/run.py", line 44, in main
    observations = env.reset()  # noqa: F841
  File "/Users/jacobclaessens/Desktop/code/habitat-2022/habitat-lab/habitat/core/env.py", line 253, in reset
    self.reconfigure(self._config)
  File "/Users/jacobclaessens/Desktop/code/habitat-2022/habitat-lab/habitat/core/env.py", line 339, in reconfigure
    self._sim.reconfigure(self._config.SIMULATOR)
  File "/Users/jacobclaessens/Desktop/code/habitat-2022/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 158, in reconfigure
    super().reconfigure(config)
  File "/Users/jacobclaessens/Desktop/code/habitat-2022/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 426, in reconfigure
    super().reconfigure(self.sim_config)
  File "/opt/anaconda3/envs/hab_v1/lib/python3.9/site-packages/habitat_sim-0.2.1-py3.9-macosx-11.0-arm64.egg/habitat_sim/simulator.py", line 256, in reconfigure
    self.__set_from_config(config)
  File "/opt/anaconda3/envs/hab_v1/lib/python3.9/site-packages/habitat_sim-0.2.1-py3.9-macosx-11.0-arm64.egg/habitat_sim/simulator.py", line 260, in __set_from_config
    self._config_backend(config)
  File "/opt/anaconda3/envs/hab_v1/lib/python3.9/site-packages/habitat_sim-0.2.1-py3.9-macosx-11.0-arm64.egg/habitat_sim/simulator.py", line 207, in _config_backend
    super().reconfigure(config.sim_cfg)
AssertionError: ESP_CHECK failed: Missing (at least) one of scene dataset attributes, stage attributes, or dataset scene attributes for scene 'data/replica_cad/configs/scenes/v3_sc4_staging_16.scene_instance.json'.  Likely an invalid scene name.
[07:01:29:401065]:[Sim] Simulator.cpp(65)::~Simulator : Deconstructing Simulator
```

aclegg3, 2024-10-28:
Hey @Claessens14 and @MeltedHyperion07 
The sc4 files are unreleased "test" split scenes. You should not expect them to exist for this use case.
If you want to run the example.py code directly on ReplicaCAD, you can remove those `sc4` scene line entries from the `*.scene_dataset_config.json` file and they won't be looked for any more.

closed 2024-10-28 by aclegg3