facebookresearch/habitat-lab #2082: malloc_consolidate(): unaligned fastbin chunk detected while testing the Humanoid Avatar

EftePA, 2024-09-19:
I wanted to test [humanoid avatars ](https://github.com/facebookresearch/habitat-lab/tree/main/habitat-lab/habitat/articulated_agents/humanoids#humanoid-design), so I downloaded files inside "habitat-sim". 

Then, first, I tried the [demo](https://github.com/facebookresearch/habitat-lab/tree/main/habitat-lab/habitat/articulated_agents/humanoids#testing-the-humanoid) by:
```
python examples/interactive_play.py --never-end --disable-inverse-kinematics --control-humanoid --cfg benchmark/rearrange/play_human.yaml

P.s. I ran it inside habitat-sim, to do so, I had to change the path to .yaml, which for me is inside habitat-lab (also tried to run the command inside habitat-lab but got bad access error!)
```
Now I can see a female avatar inside an apartment that I can change the camera position by WASD, but I don't see that the arm of the avatar moves by random.  So my first question is **why the arm of the avatar doesn't move?** Did I forget some settings?

Second, I tried [Testing the Humanoid](https://github.com/facebookresearch/habitat-lab/tree/main/habitat-lab/habitat/articulated_agents/humanoids#testing-the-humanoid) by:

```
inside habitat-lab:
python -m pytest test/test_humanoid.py

P.s. I had to change "assert kin_humanoid.get_robot_sim_id() == 1" to "assert kin_humanoid.get_robot_sim_id() == 2" to get it working!
```

and I got the following message:
```
2 passed, 1 skipped, 2 warnings in 50.40s 
malloc_consolidate(): unaligned fastbin chunk detected
Aborted (core dumped)
```
So does any one know **how to solve malloc_consolidate(): unaligned fastbin chunk detected error**?

Thank you in advance!

My habitat version:
habitat-baselines  0.3.1    
habitat-lab    0.3.1    
habitat_sim    0.3.1

SkyDream666, 2025-03-23:
Hi, I have the same problem. 
I reinstall the habitat_sim: conda install habitat-sim withbullet -c conda-forge -c aihabitat 
and the installed habitat_sim version is 0.3.3.  
then it solves the problem.

bksuh, 2025-07-15:
@RobotEverywhere hello, I faced the same problem, my question is do I only have to update habitat-sim to `0.3.3` leaving `habitat-lab==0.3.1` and `habitat-baseline==0.3.1`?? or did you change lab and baselines also to `0.3.3`