facebookresearch/habitat-lab #2086: No such file or directory: 'data/datasets/pointnav/mp3d/v1/val/val.json.gz'

jingma-git, 2024-09-25:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: 0.3.1
Habitat-Sim: 0.3.1

## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:

1. https://aihabitat.org/docs/habitat-lab/habitat-lab-demo
2. run the following code
```
config = habitat.get_config(config_path="benchmark/nav/pointnav/pointnav_mp3d.yaml")
with read_write(config):
    config.habitat.dataset.split = "val"

env = habitat.Env(config=config)
```
it gives me the error

```
{
	"name": "FileNotFoundError",
	"message": "[Errno 2] No such file or directory: 'data/datasets/pointnav/mp3d/v1/val/val.json.gz'",
	"stack": "---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
Cell In[4], line 6
      3 with read_write(config):
      4     config.habitat.dataset.split = \"val\"
----> 6 env = habitat.Env(config=config)

File ~/Project/habitat-lab-stable/habitat-lab/habitat/core/env.py:88, in Env.__init__(self, config, dataset)
     86 self._dataset = dataset
     87 if self._dataset is None and config.dataset.type:
---> 88     self._dataset = make_dataset(
     89         id_dataset=config.dataset.type, config=config.dataset
     90     )
     92 self._current_episode = None
     93 self._episode_iterator = None

File ~/Project/habitat-lab-stable/habitat-lab/habitat/datasets/registration.py:22, in make_dataset(id_dataset, **kwargs)
     19 _dataset = registry.get_dataset(id_dataset)
     20 assert _dataset is not None, \"Could not find dataset {}\".format(id_dataset)
---> 22 return _dataset(**kwargs)

File ~/Project/habitat-lab-stable/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py:116, in PointNavDatasetV1.__init__(self, config)
    112     return
    114 datasetfile_path = config.data_path.format(split=config.split)
--> 116 self._load_from_file(datasetfile_path, config.scenes_dir)
    118 # Read separate file for each scene
    119 dataset_dir = os.path.dirname(datasetfile_path)

File ~/Project/habitat-lab-stable/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py:105, in PointNavDatasetV1._load_from_file(self, fname, scenes_dir)
    103         self.from_binary(pickle.load(f), scenes_dir=scenes_dir)
    104 else:
--> 105     with gzip.open(fname, \"rt\") as f:
    106         self.from_json(f.read(), scenes_dir=scenes_dir)

File ~/miniforge3/envs/habitat-sim/lib/python3.9/gzip.py:58, in open(filename, mode, compresslevel, encoding, errors, newline)
     56 gz_mode = mode.replace(\"t\", \"\")
     57 if isinstance(filename, (str, bytes, os.PathLike)):
---> 58     binary_file = GzipFile(filename, gz_mode, compresslevel)
     59 elif hasattr(filename, \"read\") or hasattr(filename, \"write\"):
     60     binary_file = GzipFile(None, gz_mode, compresslevel, filename)

File ~/miniforge3/envs/habitat-sim/lib/python3.9/gzip.py:173, in GzipFile.__init__(self, filename, mode, compresslevel, fileobj, mtime)
    171     mode += 'b'
    172 if fileobj is None:
--> 173     fileobj = self.myfileobj = builtins.open(filename, mode or 'rb')
    174 if filename is None:
    175     filename = getattr(fileobj, 'name', '')

FileNotFoundError: [Errno 2] No such file or directory: 'data/datasets/pointnav/mp3d/v1/val/val.json.gz'"
}
```

jdiazram, 2024-10-23:
HI, I had the same problem, but I change:

This line: `config = habitat.get_config(config_paths="benchmark/nav/pointnav/pointnav_mp3d.yaml")`
for: `config = habitat.get_config("/path/to/clone/repo/habitat-lab/habitat-lab/habitat/config/benchmark/nav/pointnav/pointnav_mp3d.yaml")`

And add this line here:
```
with read_write(config):
    config.habitat.dataset.split = "val"
    config.habitat.dataset.data_path = "/path/to/clone/repo/habitat-lab/data/datasets/pointnav/habitat-test-scenes/v1/{split}/{split}.json.gz" #new line
```
Works so far, but the next lines:

```
def print_scene_recur(scene, limit_output=10):
    count = 0
    for level in scene.levels:
        print(
            f"Level id:{level.id}, center:{level.aabb.center},"
            f" dims:{level.aabb.sizes}"
        )
        for region in level.regions:
            print(
                f"Region id:{region.id}, category:{region.category.name()},"
                f" center:{region.aabb.center}, dims:{region.aabb.sizes}"
            )
            for obj in region.objects:
                print(
                    f"Object id:{obj.id}, category:{obj.category.name()},"
                    f" center:{obj.aabb.center}, dims:{obj.aabb.sizes}"
                )
                count += 1
                if count >= limit_output:
                    return None

# Print semantic annotation information (id, category, bounding box details)
# for the current scene in a hierarchical fashion
scene = env.sim.semantic_annotations()
print_scene_recur(scene, limit_output=15)

env.close()
# Note: Since only one OpenGL is allowed per process,
# you have to close the current env before instantiating a new one.
```
Only this error:

```
[13:51:17:361352]:[Warning]:[Metadata] SceneDatasetAttributes.cpp(107)::addNewSceneInstanceToDataset : Dataset : 'default' : Lighting Layout Attributes 'no_lights' specified in Scene Attributes but does not exist in dataset, so creating default.
[13:51:17:361426]:[Warning]:[Scene] SemanticScene.h(328)::checkFileExists : ::loadSemanticSceneDescriptor: Filedata/scene_datasets/habitat-test-scenes/van-gogh-room.scndoes not exist.  Aborting load.
[13:51:17:361436]:[Warning]:[Scene] SemanticScene.cpp(121)::loadSemanticSceneDescriptor : SSD File Naming Issue! Neither SemanticAttributes-provided name : `data/scene_datasets/habitat-test-scenes/van-gogh-room.scn` nor constructed filename : `data/scene_datasets/habitat-test-scenes/info_semantic.json` exist on disk.
[13:51:17:361441]:[Error]:[Scene] SemanticScene.cpp(137)::loadSemanticSceneDescriptor : SSD Load Failure! File with SemanticAttributes-provided name `data/scene_datasets/habitat-test-scenes/van-gogh-room.scn` exists but failed to load.
[13:51:17:845084]:[Warning]:[Sim] Simulator.cpp(594)::instanceStageForSceneAttributes : The active scene does not contain semantic annotations : activeSemanticSceneID_ = 0
2024-10-23 13:51:17,846 Initializing task Nav-v0
```
Can you solve the problem above and you know about this new error? and where is:"data/scene_datasets/habitat-test-scenes/van-gogh-room.scn"?

Gtrees-w, 2024-11-21:
Hi I also met the same issue.where is scn?
Have you solved it?

aaalaqeel, 2025-02-11:
The code works when I run it as a .py file
Changes were:
1- remove the s at the end of "config_paths" attribute in habitat.get_config
2- Uninstalled "opencv-python" and installed "opencv-python-headless"

I am not sure how to fix it with notebook file.

RookieXwc, 2025-02-19:
@jdiazram Thank you for your help. After doing so, I encountered the same problem as you, which is the lack of `van-gogh-room.scn` or `info_demantic.json`. It seems that the lack of certain annotation files has resulted in my semantic output being empty.

[image]
@aaalaqeel Have you encountered the same problem? Is your semantic output correct? Thank you for your help!