facebookresearch/habitat-lab #2089: where is the "datasets/objectnav/hm3d_v2.yaml"?

Jay-Vim-Lv, 2024-10-10:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: vx.x.x or master?
main
Habitat-Sim: vx.x.x or master?
main
Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/
yes
Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
yes
Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help
I noticed that datasets/objectnav/hm3d_v2.yaml for object goal navigation is described in the dataste.md, but the link transfer to 404.
I cant find the document mentioned above? can anyone help me?

aclegg3, 2024-10-29:
Hey @Jay-Vim-Lv 

I'm not able to find any broken links.
Are you following this dataset guide for HM3d? https://github.com/facebookresearch/habitat-sim/blob/main/DATASETS.md#habitat-matterport-3d-research-dataset-hm3d

You should access the HM3D github page by registering an account with Matterport and requesting key access.

Jay-Vim-Lv, 2024-10-31:
@aclegg3 
Thanks for your reply, and i know how to download the dataset.
what i mean is the document [habitat-lab/DATASETS.md](https://github.com/facebookresearch/habitat-lab/blob/main/DATASETS.md) here.

`config to use` for the task `[Object goal navigation](https://arxiv.org/abs/2006.13171)` and scene HM3DSem-v0.2 is [datasets/objectnav/hm3d_v2.yaml](https://github.com/facebookresearch/habitat-lab/blob/main/habitat-lab/habitat/config/habitat/dataset/objectnav/hm3d_v2.yaml), but that didn't offer.

and one more question is that when i try to use the ddppo to  train the objectnav task in hm3d, the result seems not right, just like the reply in the [this](https://github.com/facebookresearch/habitat-lab/issues/1563) issue,1563, and i want to know how should i do.

tangwh20, 2025-05-14:
Same question, any solution?