facebookresearch/habitat-lab #2090: Semantics aren't loading correctly (Hm3D, Instance Image Nav)

FlightVin, 2024-10-16:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: master

Habitat-Sim: master

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/ -> Yes

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/ -> Yes

Perhaps your question is answered there. If not, carry on!

## Semantics aren't loading correctly 

For the sake of brevity, consider the following image showing the data present in the `minival` set of HM3D: 
[image]

The relevant files that have been listed in README for semantics are present. 

However, when I try to access the unique semantic labels [here (permalink)](https://github.com/FlightVin/pixel-nav-data-formation/blob/master/point_based_data_former.py#L80), I always get either 0 or 3. This is despite me setting the scene dataset correctly [here](https://github.com/FlightVin/pixel-nav-data-formation/blob/76bbd4e79dc45aa5edb3f165c0760bf94cb1c85f/utils.py#L159). 

I get the following warnings on InstanceImageNav-v1  

```
[20:32:03:831051]:[Warning]:[Metadata] SceneDatasetAttributes.cpp(107)::addNewSceneInstanceToDataset 
: Dataset : 'hm3d_annotated_basis' : Lighting Layout Attributes 'no_lights' specified in Scene Attrib
utes but does not exist in dataset, so creating default.                                             
[20:32:03:831186]:[Warning]:[Scene] SemanticScene.h(328)::checkFileExists : ::loadSemanticSceneDescri
ptor: File/scratch/vineeth.bhat/sg_habitat/data/scene_datasets/hm3d_v0.2/minival/00802-wcojb4TFT35/wc
ojb4TFT35.basis.scndoes not exist.  Aborting load.                                                   
[20:32:03:831202]:[Warning]:[Scene] SemanticScene.cpp(121)::loadSemanticSceneDescriptor : SSD File Na
ming Issue! Neither SemanticAttributes-provided name : `/scratch/vineeth.bhat/sg_habitat/data/scene_d
atasets/hm3d_v0.2/minival/00802-wcojb4TFT35/wcojb4TFT35.basis.scn` nor constructed filename : `/scrat
ch/vineeth.bhat/sg_habitat/data/scene_datasets/hm3d_v0.2/minival/00802-wcojb4TFT35/info_semantic.json
` exist on disk.                                                                                     
[20:32:03:831210]:[Error]:[Scene] SemanticScene.cpp(137)::loadSemanticSceneDescriptor : SSD Load Fail
ure! File with SemanticAttributes-provided name `/scratch/vineeth.bhat/sg_habitat/data/scene_datasets
/hm3d_v0.2/minival/00802-wcojb4TFT35/wcojb4TFT35.basis.scn` exists but failed to load.               
[20:32:04:928962]:[Warning]:[Sim] Simulator.cpp(594)::instanceStageForSceneAttributes : The active sc
ene does not contain semantic annotations : activeSemanticSceneID_ = 0                               
2024-10-16 20:32:05,059 Initializing task InstanceImageNav-v1                                        
```

Any help would be appreciated. Thanks!

FlightVin, 2024-10-16:
Adding code here (same as the links given above):

Data formation:
```py
"""
Code used to create dataset for pixel goals. Uses latest habitat lab version.
"""

import os
import argparse
import numpy as np
import habitat
from habitat.tasks.nav.shortest_path_follower import ShortestPathFollower
import time
from pathlib import Path
import h5py
from habitat.utils.geometry_utils import quaternion_from_coeff, quaternion_to_list
from utils import *

seed_everything(42)

def main(args):
    config_path = args.config_path
    if not os.path.exists(config_path):
        raise RuntimeError(f"{config_path} does not exist!")

    habitat_config = create_habitat_config(config_path, args)

    env = habitat.Env(habitat_config)
    env.seed(42)
    follower = ShortestPathFollower(env.sim, goal_radius=0.5, return_one_hot=False)

    episode_counter = 0
    time_before_loop = time.time()

    hdf5_file_path = args.save_path

    # create parent directories
    directory_path = Path(hdf5_file_path).parent
    directory_path.mkdir(parents=True, exist_ok=True)

    # Creation loop
    with h5py.File(hdf5_file_path, "w") as hdf:
        while episode_counter < args.num_saved_episodes:
            start_time = time.time()

            try:
                obs = env.reset()
            except Exception as e:
                print(f"An error occurred while resetting environment: {e}")
                continue

            rgb_data, depth_data, pose_data, action_data, normed_target_point_data = (
                [],
                [],
                [],
                [],
                [],
            )
            timesteps = 0

            (
                goal_flag,
                goal_image,
                goal_mask,
                goal_point,
                height_uncorrected_goal_point,
            ) = random_pixel_goal(habitat_config, env, args.mask_shape)
            if not goal_flag:
                print(f"Rejected the current goal with goal-flag {goal_flag}")
[... 298 lines cut ...]
      main_agent:
        sim_sensors:
          rgb_sensor:
            width: 360
            height: 640
            hfov: 42
            position: [0, 1.31, 0]
          depth_sensor:
            width: 360
            height: 640
            hfov: 58
            min_depth: 0.05
            max_depth: 50.0
            position: [0, 1.31, 0]
          semantic_sensor:
            type: HabitatSimSemanticSensor
            width: 360
            height: 640
            position: [0, 1.31, 0]
        height: 1.41
        radius: 0.17
    habitat_sim_v0:
      gpu_device_id: 0
      allow_sliding: False
```

zhouak47, 2024-10-20:
I also meet the same problem  home/zhoulei/dataset_total/habitat/scenes/hm3d_v0.2/val/00829-QaLdnwvtxbs/QaLdnwvtxbs.basis.scn exists but failed to load . Have you solved this?
[image]

FlightVin, 2024-10-20:
@zhouak47 Sorry, I haven't found a solution yet.

zhouak47, 2024-10-22:
I have found a solution . You can  comment these codes out in .../habitat-lab/habitat-lab/habitat/core/env.py  and try again:
[image]
If it does not work . Try to use habitat==0.2.5 instead of 0.3.Hope this will help you. Best wishes.

aclegg3, 2024-10-29:
Interesting, thanks for the report on this issue, folks. Logging it as a bug for someone to look into.

ByZ0e, 2024-11-08:
> ## Habitat-Lab and Habitat-Sim versions
> Habitat-Lab: master
> 
> Habitat-Sim: master
> 
> Master branch contains 'bleeding edge' code and should be used at your own risk.
> 
> ## Docs and Tutorials
> Did you read the docs? https://aihabitat.org/docs/habitat-lab/ -> Yes
> 
> Did you check out the tutorials? https://aihabitat.org/tutorial/2020/ -> Yes
> 
> Perhaps your question is answered there. If not, carry on!
> 
> ## Semantics aren't loading correctly
> For the sake of brevity, consider the following image showing the data present in the `minival` set of HM3D: ![Pasted image 20241016200316 1](https://private-user-images.githubusercontent.com/35373409/377113186-f14ea935-cc21-4e7b-8f4b-c4e188d20bec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEwMzgwNzEsIm5iZiI6MTczMTAzNzc3MSwicGF0aCI6Ii8zNTM3MzQwOS8zNzcxMTMxODYtZjE0ZWE5MzUtY2MyMS00ZTdiLThmNGItYzRlMTg4ZDIwYmVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA4VDAzNDkzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM5YWVmYWQ0ZGZhNWYzNmE3MzIxYjM5N2Y3NzZjMGJkZWQ4MjdkZjcyMjBiYTkwZjU2NmNjZWNlNDIxMDc2YzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.NLzXj1DdoudQmq-sjycMNwZv6qH24UhP7OV6QKKWNKU)
> 
> The relevant files that have been listed in README for semantics are present.
> 
> However, when I try to access the unique semantic labels [here (permalink)](https://github.com/FlightVin/pixel-nav-data-formation/blob/master/point_based_data_former.py#L80), I always get either 0 or 3. This is despite me setting the scene dataset correctly [here](https://github.com/FlightVin/pixel-nav-data-formation/blob/76bbd4e79dc45aa5edb3f165c0760bf94cb1c85f/utils.py#L159).
> 
> I get the following warnings on InstanceImageNav-v1
> 
> ```
> [20:32:03:831051]:[Warning]:[Metadata] SceneDatasetAttributes.cpp(107)::addNewSceneInstanceToDataset 
> : Dataset : 'hm3d_annotated_basis' : Lighting Layout Attributes 'no_lights' specified in Scene Attrib
> utes but does not exist in dataset, so creating default.                                             
> [20:32:03:831186]:[Warning]:[Scene] SemanticScene.h(328)::checkFileExists : ::loadSemanticSceneDescri
> ptor: File/scratch/vineeth.bhat/sg_habitat/data/scene_datasets/hm3d_v0.2/minival/00802-wcojb4TFT35/wc
> ojb4TFT35.basis.scndoes not exist.  Aborting load.                                                   
> [20:32:03:831202]:[Warning]:[Scene] SemanticScene.cpp(121)::loadSemanticSceneDescriptor : SSD File Na
> ming Issue! Neither SemanticAttributes-provided name : `/scratch/vineeth.bhat/sg_habitat/data/scene_d
> atasets/hm3d_v0.2/minival/00802-wcojb4TFT35/wcojb4TFT35.basis.scn` nor constructed filename : `/scrat
> ch/vineeth.bhat/sg_habitat/data/scene_datasets/hm3d_v0.2/minival/00802-wcojb4TFT35/info_semantic.json
> ` exist on disk.                                                                                     
> [20:32:03:831210]:[Error]:[Scene] SemanticScene.cpp(137)::loadSemanticSceneDescriptor : SSD Load Fail
> ure! File with SemanticAttributes-provided name `/scratch/vineeth.bhat/sg_habitat/data/scene_datasets
> /hm3d_v0.2/minival/00802-wcojb4TFT35/wcojb4TFT35.basis.scn` exists but failed to load.               
> [20:32:04:928962]:[Warning]:[Sim] Simulator.cpp(594)::instanceStageForSceneAttributes : The active sc
> ene does not contain semantic annotations : activeSemanticSceneID_ = 0                               
> 2024-10-16 20:32:05,059 Initializing task InstanceImageNav-v1                                        
> ```
> 
> Any help would be appreciated. Thanks!

sorry, i meet the same problem here. I use the habitat==0.2.5. And i have tried the solution in https://github.com/facebookresearch/habitat-lab/issues/2090#issuecomment-2428528485. But it did not work for me.

I got the error like this:
ESP_CHECK failed: No Stage Attributes exists for requested scene 'data/scene_datasets/hm3d_v0.2/minival/00802-wcojb4TFT35/wcojb4TFT35.basis.glb' in currently specified Scene Dataset `./data/scene_datasets/hm3d_v0.2/hm3d_annotated_basis.scene_dataset_config.json`. Likely the Scene Dataset Configuration requested was not found and so a new, empty Scene Dataset was created. Verify the Scene Dataset Configuration file name used.

Any suggestions, please.

francescotaioli, 2025-05-17:
a few notes that I found here and there that solve my issues:
1. if you have a custom dataset, inside the ```from_json``` method:
``` episode.scene_dataset_config_file = "data/scene_datasets/hm3d/hm3d_annotated_basis.scene_dataset_config.json"```
2. in your main, override the config:
```
    with read_write(config):
        config.habitat.seed = args.seed
        config.habitat.simulator.scene_dataset = "data/scene_datasets/hm3d/hm3d_annotated_basis.scene_dataset_config.json" # !important, otherwise is "default"
        config.habitat.simulator.scene_dataset_config_file = "data/scene_datasets/hm3d/hm3d_annotated_basis.scene_dataset_config.json"
```
3. once you have the semantic sensor working,
```instance_id_to_label_id = {int(obj.id.split("_")[-1]): obj.category.index() for obj in sim.semantic_scene.objects}```

```   
from habitat_sim.utils.common import d3_40_colors_rgb
def display_semantic( semantic_obs: ndarray, instance_id_to_label_id):
        # Map instance IDs to label IDs
        label_map = np.vectorize(instance_id_to_label_id.get)(semantic_obs)

        # Prepare the palette image
        semantic_img = Image.new("P", (label_map.shape[1], label_map.shape[0]))
        semantic_img.putpalette(d3_40_colors_rgb.flatten())
        semantic_img.putdata((label_map.flatten() % 40).astype(np.uint8))
        semantic_img = semantic_img.convert("RGB")
        semantic_img.save("semantic.jpg")
```
```display_semantic(semantic_obs, instance_id_to_label_id)```

shipshipliucixin1, 2025-06-10:
Thanks a lot for raising this important question(for me)

shipshipliucixin1, 2025-06-10:
> [@FlightVin](https://github.com/FlightVin) I have found a solution . You can comment these codes out in .../habitat-lab/habitat-lab/habitat/core/env.py ,check your path of habitat.simulator.scene_dataset and try again: [image] If it does not work . Try to use habitat==0.2.5 instead of 0.3.Hope this will help you. Best wishes.

For why this will help, I asked Gemini.Here is its answer:
您好！这是一个非常棒的调试发现，您的分析完全正确。您发现的这个解决方案之所以有效，其核心原理在于**解决了配置覆盖（Configuration Override）的问题**。

让我为您详细解释一下这个过程：

### 核心原理：语义配置被基础配置覆盖

简单来说，您遇到的问题可以概括为：
**一个带有完整语义信息路径的正确配置，被一个只包含基础信息、缺少语义路径的配置给覆盖了，导致模拟器无法加载语义场景。**

下面是详细的步骤分解：

#### 1. 初始状态：正确的“语义感知”配置

1.  当您在主程序 `objnav_benchmark.py` 中加载一个任务配置（比如一个YAML文件）时，这个配置通常会指定一个包含语义信息的 `scene_dataset`。例如，它可能会指向 `hm3d_semantic.scene_dataset_config.json`。
2.  这个 **“语义配置文件”** (semantic config) 是关键，它不仅告诉模拟器场景的视觉模型（`.glb`文件）在哪里，**还包含了指向语义数据（如`.semantic.json`或`.house`文件）的路径**。

#### 2. 问题所在：`env.py` 中的配置覆盖

1.  在 `habitat.Env` 的 `__init__` 方法中，程序会加载第一个 `episode` 来确定要使用的初始场景。
2.  然后，执行了这行您注意到的关键代码：
    ```python
    self._config.simulator.scene_dataset = (
        self.current_episode.scene_dataset_config
    )
    ```
3.  这里的 `self.current_episode.scene_dataset_config` 通常指向一个**更基础的配置文件**，比如 `hm3d_basis.scene_dataset_config.json`。这个 **“基础配置文件”** 只包含了运行基本导航任务所需的最少信息，但**它没有包含加载语义场景所需的路径**。
4.  **结果**：这行代码将您原来正确、完整的“语义配置”用一个不完整的“基础配置”给覆盖了。

#### 3. 产生错误：模拟器加载失败

1.  当模拟器（`HabitatSim`）根据这个被覆盖后的“基础配置”进行初始化或重置时，它依然会尝试加载语义信息（因为您的任务需要它）。
2.  但由于“基础配置”中没有提供正确的语义文件路径，模拟器只能尝试根据默认规则或不完整的信息去猜测路径。
3.  这就导致了您看到的 `SSD Load Failure! ... .basis.scn` 错误。模拟器尝试加载一个它认为应该存在的语义文件，但该文件要么不存在，要么格式不正确，最终导致加载失败。
4.  加载失败的直接后果就是 `self._sim.semantic_annotations().objects` 返回一个空列表，因为没有任何语义对象被成功加载。

### 您的解决方案为何有效？

通过注释掉这行代码：
```python
# self._config.simulator.scene_dataset = (
#     self.current_episode.scene_dataset_config
# )
```

您阻止了配置覆盖的发生。

1.  **保留了原始配置**：模拟器现在会一直使用您在任务级YAML文件中指定的、**完整且正确的“语义配置”**。
2.  **成功加载语义**：因为这个配置包含了正确的语义文件路径，`HabitatSim` 在初始化时可以成功找到并加载 `*.semantic.json` 文件。
3.  **问题解决**：`self._sim.semantic_annotations().objects` 因此能被正确填充，后续代码（如 `_draw_goals_aabb`）在访问 `sem_scene.objects[object_id]` 时就不会再出错。

### 总结与类比

您可以把这个过程想象成准备一次旅行：

*   **您的主配置文件**：是一份完整的旅行指南，上面有地图、景点介绍（语义信息）和酒店地址。
*   **Episode里的配置**：是一张只包含地图的简易传单。
*   **原始代码的行为**：强制您扔掉完整的旅行指南，只拿着那张简易传单出发。当您想找某个景点（语义对象）的详细介绍时，自然就找不到了。
*   **您的解决方案**：就是决定不理会那张简易传单，一直使用您最初那本完整的旅行指南。

因此，对于需要场景语义的任务（如ObjectGoal Navigation），您采取的这个修改是完全正确且必要的。这行代码的本意可能是为了支持那些不需要语义，或者每个episode都自带完整语义路径的简单任务。

liliang18273158269, 2025-06-11:
Hi,@shipshipliucixin1 ,请问你的意思是注释掉：
self._config.simulator.scene_dataset = (
    self.current_episode.scene_dataset_config
)
这部分内容就可以吗

shipshipliucixin1, 2025-06-11:
> Hi,[@shipshipliucixin1](https://github.com/shipshipliucixin1) ,请问你的意思是注释掉： self._config.simulator.scene_dataset = ( self.current_episode.scene_dataset_config ) 这部分内容就可以吗，需不需要进行类似如下改动呢： with read_write(config): config.habitat.seed = args.seed config.habitat.simulator.scene_dataset = "data/scene_datasets/hm3d/hm3d_annotated_basis.scene_dataset_config.json" config.habitat.simulator.scene_dataset_config_file ="data/scene_datasets/hm3d/hm3d_annotated_basis.scene_dataset_config.json"

是的，在habitat-lab这个库里面，只需要注释掉这个有bug的代码就行。因为这句话把json.gz文件里的错误的路径拿过来，覆盖了我们在外面定义的正确的路径。但是在habitat-lab这个库以外我们自己代码的里面，得写上这些内容