facebookresearch/habitat-lab #2093: How to use HumanoidDetector to detect specific humanoid in image?

wsakobe, 2024-10-24:
Hi, I am currently working on creating a social navigation task involving multiple humanoids using Habitat 3.0. At the moment, I need to configure a lab sensor, primarily to detect the presence of a **specific humanoid** in an image (similar to the HumanoidDetectorSensor in Habitat’s native social_nav_sensors). However, I’ve encountered a few issues:

1. First, what is the main idea of this detector? From what I observe, it doesn’t seem to process the input image but simply checks whether the image pixels match a certain human_id. Is this approach reasonable, or am I missing some important details?

2. Secondly, how is human_id defined and utilized? If I have six humanoids in my scene (agent_0, agent_1, … agent_6), how can I configure or modify the detector to specifically detect the presence or absence of **agent_0**?

Thank you in advance. I look forward to your reply!

aclegg3, 2024-10-29:
Hey @wsakobe,

Quick answers and let me know if you have further questions:

1. >what is the main idea of this (humanoid) detector? ... simply checks whether the image pixels match a certain human_id.

That is correct. The sensor is measuring whether a robot can see a human or not. That signal is expected to be something a real-world, off the shelf, computer vision model may be able to provide. Anything else is left up the nav policy.

2. > how is human_id defined and utilized? 
Relevant: https://github.com/facebookresearch/habitat-lab/issues/2045
Human ids are defined in the `ao_config.json` for the assets. They can also be re-defined after loading the humanoid in the simulator if necessary. 
> how can I configure or modify the detector to specifically detect the presence or absence of agent_0?
One option is to manually set each humanoid asset to a different `semantic_id` in advance. Another option is to use a `panoptic sensor` which provides instance segmentation (rendering the `object_id` of the humanoid's `ManagedArticulatedObject`). Then you could check for the specific `object_ids` you are looking for.

wsakobe, 2024-10-30:
Hi @aclegg3 ,
I have no further questions at the moment. Thank you again for your valuable response!

closed 2024-10-30 by wsakobe