facebookresearch/habitat-lab #2101: Observation height

realjoshqsun, 2024-11-04:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: v0.1.7

Habitat-Sim: v0.1.7

Hi, I want to change the height of observation to simulate drone flying height or robocar observation height. Any ideas?

aclegg3, 2024-11-04:
The Agent configs define the offset placement of the standard visual sensors: https://github.com/facebookresearch/habitat-lab/blob/2c164c3a29d1d4eabcb3d183479e9696ed002515/habitat-lab/habitat/config/default_structured_configs.py#L1665

realjoshqsun, 2024-11-04:
@aclegg3 
Hi, thank you so much.

I was wondering if this height setting can be added in some config files like
ENVIRONMENT:
  MAX_EPISODE_STEPS: 500
SIMULATOR:
  ACTION_SPACE_CONFIG: v0
  AGENT_0:
    SENSORS: [RGB_SENSOR, DEPTH_SENSOR]
  FORWARD_STEP_SIZE: 0.25
  TURN_ANGLE: 15
  HABITAT_SIM_V0:
    GPU_DEVICE_ID: 0
    ALLOW_SLIDING: True
  RGB_SENSOR:
    WIDTH: 224
    HEIGHT: 224
    HFOV: 90
    TYPE: HabitatSimRGBSensor
  DEPTH_SENSOR:
    WIDTH: 256
    HEIGHT: 256
TASK:
  TYPE: VLN-v0
  SUCCESS_DISTANCE: 3.0
  SENSORS: [
    INSTRUCTION_SENSOR,
    SHORTEST_PATH_SENSOR,
    VLN_ORACLE_PROGRESS_SENSOR
  ]
  INSTRUCTION_SENSOR_UUID: instruction
  POSSIBLE_ACTIONS: [STOP, MOVE_FORWARD, TURN_LEFT, TURN_RIGHT]
  MEASUREMENTS: [
    DISTANCE_TO_GOAL,
    SUCCESS,
    SPL,
    NDTW,
    PATH_LENGTH,
    ORACLE_SUCCESS,
    STEPS_TAKEN
  ]
  SUCCESS:
    SUCCESS_DISTANCE: 3.0
  SPL:
    SUCCESS_DISTANCE: 3.0
  NDTW:
    GT_PATH: data/datasets/R2R_VLNCE_v1-3_preprocessed/{split}/{split}_gt.json.gz
DATASET:
  TYPE: VLN-CE-v1
  SPLIT: train
  DATA_PATH: data/datasets/R2R_VLNCE_v1-3_preprocessed/{split}/{split}.json.gz
  SCENES_DIR: data/scene_datasets/
  
  Or it's hard coded in habitat-lab and the only way is to change the source code before building habitat.

aclegg3, 2024-11-04:
Hey @realjoshqsun,

Yeah, you should be able to set that here under `AGENT_0:` Anything in the default config file I linked can be set in the YAML. The section I linked are the options under the Agent's config.

aclegg3, 2024-11-04:
Check out the configs from the previous navigation challenge: https://github.com/facebookresearch/habitat-challenge/blob/main/configs/benchmark/nav/imagenav/imagenav_hm3d_v3_challenge.yaml

You can see here that the individual sensors are getting offset adjustments for where they attach to the agent.

realjoshqsun, 2024-11-04:
Thank you for your quick response. I really appreciate your help.

aclegg3, 2024-11-04:
Closing this issue since the question seems to be resolved. Feel free to open another if you have more questions.

closed 2024-11-04 by aclegg3

realjoshqsun, 2024-11-27:
@aclegg3 

Hi, I tried to change config file like below:

SIMULATOR:
  ACTION_SPACE_CONFIG: v0
  AGENT_0:
    SENSORS: [RGB_SENSOR, DEPTH_SENSOR]
    HEIGHT: 2.0
    
I think it should work because some information has been displayed through logger:
```
I1126 20:38:05.853909 2464894 simulator.py:213] Loaded navmesh data/scene_datasets/mp3d/JeFG25nYj2p/JeFG25nYj2p.navmesh
I1126 20:38:05.854743 2464894 simulator.py:224] Recomputing navmesh for agent's height 2.0 and radius 0.1.
I1126 20:38:05.860105 2464894 PathFinder.cpp:382] Building navmesh with 484x365 cells
I1126 20:38:06.004103 2464894 PathFinder.cpp:650] Created navmesh with 718 vertices 349 polygons
I1126 20:38:06.004148 2464894 Simulator.cpp:710] reconstruct navmesh successful
2024-11-26 20:38:06,006 Initializing task VLN-v0
2024-11-26 20:38:13,333 Agent parameters: 57945206. Trainable: 2457477
2024-11-26 20:38:13,334 Finished setting up policy.
2024-11-26 20:38:13,336 Initializing dataset VLN-CE-v1
I1126 20:38:13.999364 2464894 PhysicsManager.cpp:33] Deconstructing PhysicsManager
I1126 20:38:13.999421 2464894 SemanticScene.h:41] Deconstructing SemanticScene
I1126 20:38:14.002642 2464894 SceneManager.h:25] Deconstructing SceneManager
I1126 20:38:14.002652 2464894 SceneGraph.h:26] Deconstructing SceneGraph
I1126 20:38:14.003418 2464894 Sensor.h:87] Deconstructing Sensor
I1126 20:38:14.003448 2464894 Sensor.h:87] Deconstructing Sensor
I1126 20:38:14.013032 2464894 Renderer.cpp:34] Deconstructing Renderer
I1126 20:38:14.013059 2464894 WindowlessContext.h:17] Deconstructing WindowlessContext
I1126 20:38:14.015496 2464894 Simulator.cpp:49] Deconstructing Simulator
```

However, the output is still the same when I do
```output = envs.step(action)```