facebookresearch/habitat-lab #2103: How To Use Continuous Action Space For ObjectNav?

AlexandreBrown, 2024-11-05:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: 0.3.2

Habitat-Sim: 0.3.2

## ❓ Questions and Help
Hi, I would like to know what is a minimally working example of using continuous action space for ObjectNav ? I tried editing the habitat/task/objectnav.yaml without success, my env action space remains discrete.  

Starting code :  
```python
#!/usr/bin/env python3

# Copyright (c) Meta Platforms, Inc. and its affiliates.
# This source code is licensed under the MIT license found in the
# LICENSE file in the root directory of this source tree.

import gym

import habitat.gym  # noqa: F401
import cv2

def example():
    config = habitat.get_config(
        "benchmark/nav/objectnav/objectnav_hssd-hab.yaml"
    )
    env = habitat.gym.make_gym_from_config(config)

    print("Environment creation successful")
    observations = env.reset()  # noqa: F841
    rgb = observations["rgb"]

    height, width, channels = rgb.shape
    video_writer = cv2.VideoWriter(
        "recorded_video.mp4",
        cv2.VideoWriter_fourcc(*"mp4v"),
        30,
        (width, height),
    )

    print("Agent acting inside environment.")
    count_steps = 0
    terminal = False
    while not terminal:
        rgb = observations["rgb"]
        video_writer.write(cv2.cvtColor(rgb, cv2.COLOR_RGB2BGR))

        observations, reward, terminal, info = env.step(
            env.action_space.sample()
        )  # noqa: F841
        count_steps += 1

    env.close()
    video_writer.release()
    print("Episode finished after {} steps.".format(count_steps))

if __name__ == "__main__":
    example()

```

AlexandreBrown, 2024-11-05:
CC @0mdc @aclegg3

tangwh20, 2025-03-25:
Same problem. Any conclusion?