facebookresearch/habitat-lab #2104: GL::Context: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidVal

linyx700, 2024-11-06:
(habitat)` lyx@lyxHP:~/habitat$ python habitat-sim/examples/example.py
sim_cfg.physics_config_file = data/default.physics_config.json
GL::Context: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidValue

is there anything wrong about the nvidia driver?
[image]

closed 2024-11-19 by linyx700

lasuomela, 2024-11-20:
Hi,

Did you manage to fix this? Running into the same issue.

chuangguo, 2024-11-28:
The same issue, how did you solve it?

yaioqin, 2024-12-07:
I downgraded the version of Habitat to 0.2.5, and the error is gone.

ghubnerr, 2025-01-12:
> I downgraded the version of Habitat to 0.2.5, and the error is gone.

This worked or me -- although a little unideal to downgrade from Habitat 3. If anyone is able to fix this without the downgrade option, please let us know.

Lazy-boy193, 2025-03-02:
> For me, the issue was caused by building habitat-sim from scratch in a conda environment with torchvision installed.
> 
> By default, conda might install build versions of torchvision dependencies (opencv, ffmpeg, libva) that install libgl and libegl from conda-forge into the environment. These seem to cause a driver conflict for habitat.
> 
> Installing these versions/builds solved the issue for me: [image]

How to find these packages and install them? I couldn't find a specific version in the conda installation. Furthermore, reducing the habitat to 0.2.5 cannot solve this problem: GL::Context: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidValue

lasuomela, 2025-03-02:
> How to find these packages and install them? I couldn't find a specific version in the conda installation. Furthermore, reducing the habitat to 0.2.5 cannot solve this problem: GL::Context: cannot [...]

Hello,

My example image was from the conda environment.yml file. Below is my complete environment for reference. You might need to change the specific build of libva to fit your platform, but it has to be one that does not depend on libgl, libegl, or libglvnd.

```yaml
name: habitat_py310
channels:
  - pytorch
  - nvidia
  - huggingface
  - aihabitat
  - defaults
  - conda-forge
dependencies:
  - python=3.10
  - cmake
  - pytorch==2.4.0
  - torchvision
  - pytorch-cuda=12.4
  - ffmpeg=7.1.0
  - opencv[build=headless*] # Avoid installing libgl, libegl, libglvnd which break habitat-sim
  - libva[build=hb711507_0] # Avoid installing libgl, libegl, libglvnd
  - scipy
  - quaternion
  - numpy>=1.20.0,<1.24.0
  - numba
  - attrs
  - imageio
  - git
  - jsonpickle
  - pip
```

Lazy-boy193, 2025-03-03:
> > > For me, the issue was caused by building habitat-sim from scratch in a conda environment with torchvision installed.对我来说，该问题是由于在安装了 torchvision 的 conda 环境中从头开始构建 habitat-sim 引起的。
> > > By default, conda might install build versions of torchvision dependencies (opencv, ffmpeg, libva) that install libgl and libegl from conda-forge into the environment. These seem to cause a driver conflict for habitat.默认情况下，conda 可能会安装 torchvision 依赖项（opencv、ffmpeg、libva）的构建版本，这些版本会将 conda-forge 中的 libgl 和 libegl 安装到环境中。这似乎会导致 habitat 驱动程序冲突。
> > > Installing these versions/builds solved the issue for me: 安装这些版本/构建解决了我的问题：[image]
> > 
> > 
> > How to find these packages and install them? I couldn't find a specific version in the conda installation. Furthermore, reducing the habitat to 0.2.5 cannot solve this problem: GL::Context: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidValue如何找到这些包并安装它们？我在 conda 安装中找不到特定版本。此外，将环境降低到 0.2.5 也无法解决这个问题：GL::Context：无法检索 OpenGL 版本：GL::Renderer::Error::InvalidValue
> 
> Hello,  你好，
> 
> My example image was from the conda environment.yml file. Below is my complete environment for reference. You might need to change the specific build of libva to fit your platform, but it has to be one that does not depend on libgl, libegl, or libglvnd.我的示例图像来自 conda environment.yml 文件。下面是我的完整环境，供参考。您可能需要更改 libva 的特定版本以适合您的平台，但它必须是不依赖于 libgl、libegl 或 libglvnd 的版本。
> 
> name: habitat_py310
> channels:
>   - pytorch
>   - nvidia
>   - huggingface
>   - aihabitat
>   - defaults
>   - conda-forge
> dependencies:
>   - python=3.10
>   - cmake
>   - pytorch==2.4.0
>   - torchvision
>   - pytorch-cuda=12.4
>   - ffmpeg=7.1.0
>   - opencv[build=headless*] # Avoid installing libgl, libegl, libglvnd which break habitat-sim
>   - libva[build=hb711507_0] # Avoid installing libgl, libegl, libglvnd
>   - scipy
>   - quaternion
>   - numpy>=1.20.0,<1.24.0
>   - numba
>   - attrs
>   - imageio
>   - git
>   - jsonpickle
>   - pip

Hello,I created a virtual environment using your provided environment configuration file (`habitat_environment.yml`), but I changed the Python version to 3.9. After creating the environment, I activated it using:

```bash
conda env create -f habitat_environment.yml
conda activate habitat_py39
```

Then, I tried to install `habitat-sim` version 0.3.1 using `conda` with the following command:

```bash
conda install habitat-sim=0.3.1 headless -c conda-forge -c aihabitat
```

However, I encountered the following error:

```
Channels:
 - conda-forge
 - aihabitat
 - defaults
 - nvidia
 - huggingface
 - pytorch
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - package ffmpeg-7.1.0-gpl_ha1cb39d_715 requires libxcb >=1.17.0,<2.0a0, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ ffmpeg 7.1.0**  is installable with the potential options
│  ├─ ffmpeg 7.1.0 would require
│  │  └─ libxcb >=1.17.0,<2.0a0 , which can be installed;
│  └─ ffmpeg 7.1.0 would require
│     └─ libxcb >=1.16,<1.17.0a0 , which can be installed;
├─ habitat-sim 0.3.1**  is installable with the potential options
│  ├─ habitat-sim 0.3.1 would require
│  │  └─ libxcb >=1.15,<1.16.0a0 , which conflicts with any installable versions previously reported;
│  └─ habitat-sim 0.3.1 would require
│     └─ headless <0 , which can be installed;
└─ headless is not installable because it conflicts with any installable versions previously reported.
```
So I would like to ask how you proceed with the installation of Habitat-Sim and Habitat-Lab. Thank you very much!

Lazy-boy193, 2025-03-03:
> > > For me, the issue was caused by building habitat-sim from scratch in a conda environment with torchvision installed.
> > > By default, conda might install build versions of torchvision dependencies (opencv, ffmpeg, libva) that install libgl and libegl from conda-forge into the environment. These seem to cause a driver conflict for habitat.
> > > Installing these versions/builds solved the issue for me: [image]
> > 
> > 
> > How to find these packages and install them? I couldn't find a specific version in the conda installation. Furthermore, reducing the habitat to 0.2.5 cannot solve this problem: GL::Context: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidValue
> 
> Hello,
> 
> My example image was from the conda environment.yml file. Below is my complete environment for reference. You might need to change the specific build of libva to fit your platform, but it has to be one that does not depend on libgl, libegl, or libglvnd.
> 
> name: habitat_py310
> channels:
>   - pytorch
>   - nvidia
>   - huggingface
>   - aihabitat
>   - defaults
>   - conda-forge
> dependencies:
>   - python=3.10
>   - cmake
>   - pytorch==2.4.0
>   - torchvision
>   - pytorch-cuda=12.4
>   - ffmpeg=7.1.0
>   - opencv[build=headless*] # Avoid installing libgl, libegl, libglvnd which break habitat-sim
>   - libva[build=hb711507_0] # Avoid installing libgl, libegl, libglvnd
>   - scipy
>   - quaternion
>   - numpy>=1.20.0,<1.24.0
>   - numba
>   - attrs
>   - imageio
>   - git
>   - jsonpickle
>   - pip

Hello,I created a virtual environment using your provided environment configuration file (`habitat_environment.yml`), but I changed the Python version to 3.9. After creating the environment, I activated it using:

```bash
conda env create -f habitat_environment.yml
conda activate habitat_py39
```

Then, I tried to install `habitat-sim` version 0.3.1 using `conda` with the following command:

```bash
conda install habitat-sim=0.3.1 headless -c conda-forge -c aihabitat
```

However, I encountered the following error:

[image]

So I would like to ask how you proceed with the installation of Habitat-Sim and Habitat-Lab. Thank you very much!

lasuomela, 2025-03-03:
> > > > For me, the issue was caused by building habitat-sim from scratch in a conda environment with torchvision installed.
> > > > By default, conda might install build versions of torchvision dependencies (opencv, ffmpeg, libva) that install libgl and libegl from conda-forge into the environment. These seem to cause a driver conflict for habitat.
> > > > Installing these versions/builds solved the issue for me: [image]
> > > 
> > > 
> > > How to find these packages and install them? I couldn't find a specific version in the conda installation. Furthermore, reducing the habitat to 0.2.5 cannot solve this problem: GL::Context: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidValue
> > 
> > 
> > Hello,
> > My example image was from the conda environment.yml file. Below is my complete environment for reference. You might need to change the specific build of libva to fit your platform, but it has to be one that does not depend on libgl, libegl, or libglvnd.
> > name: habitat_py310
> > channels:
> > 
> > * pytorch
> > * nvidia
> > * huggingface
> > * aihabitat
> > * defaults
> > * conda-forge
> >   dependencies:
> > * python=3.10
> > * cmake
> > * pytorch==2.4.0
> > * torchvision
> > * pytorch-cuda=12.4
> > * ffmpeg=7.1.0
> > * opencv[build=headless*] # Avoid installing libgl, libegl, libglvnd which break habitat-sim
> > * libva[build=hb711507_0] # Avoid installing libgl, libegl, libglvnd
> > * scipy
> > * quaternion
> > * numpy>=1.20.0,<1.24.0
> > * numba
> > * attrs
> > * imageio
> > * git
> > * jsonpickle
> > * pip
> 
> Hello,I created a virtual environment using your provided environment configuration file (`habitat_environment.yml`), but I changed the Python version to 3.9. After creating the environment, I activated it using:
> 
> conda env create -f habitat_environment.yml
> conda activate habitat_py39
> 
> Then, I tried to install `habitat-sim` version 0.3.1 using `conda` with the following command:
> 
> conda install habitat-sim=0.3.1 headless -c conda-forge -c aihabitat
> 
> However, I encountered the following error:
> 
> [image]
> 
> So I would like to ask how you proceed with the installation of Habitat-Sim and Habitat-Lab. Thank you very much!

Sorry, I had accidentally dropped the pip commands for installing Habitat. I edited them into my previous post. But if you just want to use Habitat with Python 3.9, you should be able to install it with conda following the instructions in the [repo](https://github.com/facebookresearch/habitat-sim). My example is for building Habitat from source, which you will need if you want to use some other version of Python e.g. 3.10.

erjiaxiao, 2025-08-06:
> For me, the issue was caused by building habitat-sim from scratch in a conda environment with torchvision installed.
>
> By default, conda might install build versions of torchvision dependencies [...]

@lasuomela Thank you so much for sharing this solution! This is incredibly helpful and saved me a lot of debugging time. I'm curious about your troubleshooting process - how did you manage to pinpoint that the issue was specifically caused by the conda-forge versions of libgl and libegl conflicting with the system drivers? This seems like a very subtle dependency conflict that would be challenging to diagnose. 

Thanks again for your solution!

lasuomela, 2025-08-06:
> [@lasuomela](https://github.com/lasuomela) Thank you so much for sharing this solution! This is incredibly helpful and saved me a lot of debugging time. I'm curious about your troubleshooting [...]

I just went back to a minimal set of dependencies and started adding them back one by one. If I remember correctly, comparing to the set of dependencies that come with the conda install of python 3.9 Habitat helped identifying libgl and libegl as a possible cause.

erjiaxiao, 2025-08-06:
@lasuomela Thank you for sharing both the solution and the investigative process behind it!

Yang-jaemin, 2025-08-27:
I also spent a lot of time struggling with a similar error, but I was able to resolve it with the following steps.

---

### ✅ Step 1: Check your current OpenGL renderer

```bash
glxinfo | grep "OpenGL renderer"
```

In my case, even though I’m using an **NVIDIA GeForce RTX 3080 Ti Laptop GPU**, the OpenGL renderer was set to **Intel integrated graphics**.

---

### ✅ Step 2: Add this environment variable to `.bashrc`

```bash
echo 'export __GLX_VENDOR_LIBRARY_NAME=nvidia' >> ~/.bashrc
source ~/.bashrc
```

After applying the change, the renderer correctly switched to the NVIDIA GPU:

```bash
glxinfo | grep "OpenGL renderer"
OpenGL renderer string: NVIDIA GeForce RTX 3080 Ti Laptop GPU/PCIe/SSE2
```

---

### ❌ Error Resolved

I no longer see the frustrating error:

```
GL::Context: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidVal
```

---

### 🧩 Environment Versions

- `habitat-sim==0.3.1`
- `habitat-lab==0.3.1`

---

Hope this helps anyone facing the same issue! 🚀