facebookresearch/habitat-lab #2105: Social Navigation Baselines: Object with handle '004_sugar_box_:0000' could not be found in the scene. Has the Episode been initialized

licongfei001, 2024-11-08:
## Habitat-Lab and Habitat-Sim versions
Habitat-Lab: 0.3.2?

Habitat-Sim: 0.3.2?

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest release versions of [Habitat-Lab](https://github.com/facebookresearch/habitat-lab/releases) and [Habitat-Sim](https://github.com/facebookresearch/habitat-sim/releases)? Your question may already be addressed in the latest versions. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-lab/
yes
Did you check out the tutorials? https://aihabitat.org/tutorial/2020/
yes

My system is ubuntu 20.04.
The run command is 
python -u -m habitat_baselines.run \
    --config-name=social_nav/social_nav.yaml \
    benchmark/multi_agent=hssd_spot_human_social_nav \
    habitat_baselines.evaluate=False \
    habitat_baselines.num_checkpoints=500 \
    habitat_baselines.total_num_steps=1.0e9 \
    habitat_baselines.num_environments=8 \
    habitat_baselines.tensorboard_dir= \
    habitat_baselines.video_dir=video_dir \
    habitat_baselines.checkpoint_folder=data/checkpoints_social_nav \
    habitat_baselines.eval_ckpt_path_dir=data/checkpoints/social_nav_latest.pth \
    habitat.task.actions.agent_0_base_velocity.longitudinal_lin_speed=10.0 \
    habitat.task.actions.agent_0_base_velocity.ang_speed=10.0 \
    habitat.task.actions.agent_0_base_velocity.allow_dyn_slide=True \
    habitat.task.actions.agent_0_base_velocity.enable_rotation_check_for_dyn_slide=False \
    habitat.task.actions.agent_1_oracle_nav_randcoord_action.lin_speed=10.0 \
    habitat.task.actions.agent_1_oracle_nav_randcoord_action.ang_speed=10.0 \
    habitat.task.actions.agent_1_oracle_nav_action.lin_speed=10.0 \
    habitat.task.actions.agent_1_oracle_nav_action.ang_speed=10.0 \
    habitat.task.measurements.social_nav_reward.facing_human_reward=3.0 \
    habitat.task.measurements.social_nav_reward.count_coll_pen=0.01 \
    habitat.task.measurements.social_nav_reward.max_count_colls=-1 \
    habitat.task.measurements.social_nav_reward.count_coll_end_pen=5 \
    habitat.task.measurements.social_nav_reward.use_geo_distance=True \
    habitat.task.measurements.social_nav_reward.facing_human_dis=3.0 \
    habitat.task.measurements.social_nav_seek_success.following_step_succ_threshold=400 \
    habitat.task.measurements.social_nav_seek_success.need_to_face_human=True \
    habitat.task.measurements.social_nav_seek_success.use_geo_distance=True \
    habitat.task.measurements.social_nav_seek_success.facing_threshold=0.5 \
    habitat.task.lab_sensors.humanoid_detector_sensor.return_image=True \
    habitat.task.lab_sensors.humanoid_detector_sensor.is_return_image_bbox=True \
    habitat.task.success_reward=10.0 \
    habitat.task.end_on_success=True \
    habitat.task.slack_reward=-0.1 \
    habitat.environment.max_episode_steps=50 \
    habitat.simulator.kinematic_mode=True \
    habitat.simulator.ac_freq_ratio=4 \
    habitat.simulator.ctrl_freq=120 \
    habitat.simulator.agents.agent_0.joint_start_noise=0.0

The errors are, 
2024-11-08 12:41:59,529 Cannot find active receptacle fed2a84682713eabe2b8d0e1e950d891c7442d5f_:0001|receptacle_mesh_fed2a84682713eabe2b8d0e1e950d891c7442d5f.0001, so cannot create a parent relationship. Skipping. Note that episode is likely invalid.
2024-11-08 12:41:59,589 Cannot find active receptacle 3b134d45ea8f2bd418e77c0d6a861dfe728f38a8_:0000|receptacle_mesh_3b134d45ea8f2bd418e77c0d6a861dfe728f38a8.0000, so cannot create a parent relationship. Skipping. Note that episode is likely invalid.
Exception in thread Thread-7:
Traceback (most recent call last):
  File "/home/cfli/anaconda3/envs/habitat/lib/python3.9/threading.py", line 980, in _bootstrap_inner
    self.run()
  File "/home/cfli/anaconda3/envs/habitat/lib/python3.9/threading.py", line 917, in run
    self._target(*self._args, **self._kwargs)
  File "/home/cfli/habitat-lab/habitat-lab/habitat/core/vector_env.py", line 268, in _worker_env
    observations = env.reset()
  File "/home/cfli/habitat-lab/habitat-lab/habitat/gym/gym_env_episode_count_wrapper.py", line 61, in reset
    return self.env.reset(**kwargs)
[... 154 lines cut ...]
  File "/home/cfli/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 704, in _add_objs
    return func(*args, **kwds)
  File "/home/cfli/habitat-lab/habitat-lab/habitat/core/env.py", line 425, in reset
    self.kinematic_relationship_manager.initialize_from_obj_to_rec_pairs(
  File "/home/cfli/habitat-lab/habitat-lab/habitat/sims/habitat_simulator/kinematic_relationship_manager.py", line 212, in initialize_from_obj_to_rec_pairs
    observations = self._env.reset()    assert (

AssertionError  File "/home/cfli/habitat-lab/habitat-lab/habitat/core/env.py", line 261, in reset
: Object with handle '004_sugar_box_:0000' could not be found in the scene. Has the Episode been initialized?
    self.reconfigure(self._config)
  File "/home/cfli/habitat-lab/habitat-lab/habitat/core/env.py", line 343, in reconfigure
    self._sim.reconfigure(self._config.simulator, self.current_episode)
  File "/home/cfli/habitat-lab/habitat-lab/habitat/tasks/rearrange/utils.py", line 693, in wrapper
    ret = f(self, *args, **kwargs)
  File "/home/cfli/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 358, in reconfigure
    self._add_objs(ep_info, should_add_objects, new_scene)
  File "/home/cfli/habitat-lab/habitat-lab/habitat/tasks/rearrange/utils.py", line 693, in wrapper
    ret = f(self, *args, **kwargs)
  File "/home/cfli/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 704, in _add_objs
    self.kinematic_relationship_manager.initialize_from_obj_to_rec_pairs(
  File "/home/cfli/habitat-lab/habitat-lab/habitat/sims/habitat_simulator/kinematic_relationship_manager.py", line 212, in initialize_from_obj_to_rec_pairs
    assert (
AssertionError: Object with handle '004_sugar_box_:0000' could not be found in the scene. Has the Episode been initialized?

I have downloaded the dataset. Could you help me out?

iyool, 2025-04-17:
Hi @licongfei001 I am facing a similar error when trying to run the social rearrange/pop_play.yaml training task. May I know if you manage it resolve the error?

yang-ze-kang, 2025-05-15:
There has some bug in their provided episodes datat. I solved this by the following script:
```python
import json
import gzip
import copy
from collections import defaultdict
import shutil

def process(path):
    shutil.copy(path,path.replace('.json.gz','-old.json.gz'))
    with gzip.open(path,'rt') as f:
        ds = json.load(f)

    ds_new = copy.deepcopy(ds)

    for i, e in enumerate(ds['episodes']):
        if i<=1:
            continue
        objs = []
        obj2num = defaultdict(int)
        for obj in e['rigid_objs']:
            obj_name = obj[0].split('.')[0]
            objs.append(obj_name+f"_:{str(obj2num[obj_name]).zfill(4)}")
            obj2num[obj_name]+=1
        for obj in e['name_to_receptacle'].keys():
            if obj not in objs:
                del ds_new['episodes'][i]['name_to_receptacle'][obj]

    with gzip.open(path, 'wt') as f:
        json.dump(ds_new, f)

path = 'data/datasets/hssd/rearrange/train/social_rearrange.json.gz'
process(path)
path = 'data/datasets/hssd/rearrange/val/social_rearrange.json.gz'
process(path)
```